basler/pypylon #575: grabResult.GrabSucceeded() not catching unsuccesfull frames

IgnacioEscorz, 2023-01-04:
Hello everyone,
I have a code getting images from an GigE camera using the converter.Convert(grabResult), but due to what it seems bandwidth issues, sometimes the frame is not fully captured but the method grabResult.GrabSucceeded() still yields TRUE, so it might not be catching all cases of this method? Or how this method specifically works?
Thanks in advance.

from the terminal:
return _pylon.ImageFormatConverter_Convert(self, *args)
_genicam.InvalidArgumentException: Cannot convert image. The passed source image is invalid. : InvalidArgumentException thrown (file 'ImageFormatConverter.cpp', line 77)

thiesmoeller, 2023-01-04:
If you check your image _before_ the image format converter one reason could be
e.g. do you use a packetsize > 1500 Byte and a very old realtek NIC ?  ( They had a DMA issue on larger packets )

If grabsucceded is TRUE but your image is corrupt:

  it is still ok on the wire, but gets corrupted during transfer to memory

To your message from convert:
You get it when the image buffer you transfer ( it is part of the GrabResult ) is either not existing or not filled with any data.

It would help if you can post part of your code to assist in debugging.

IgnacioEscorz, 2023-01-05:
> If you check your image _before_ the image format converter one reason could be e.g. do you use a packetsize > 1500 Byte and a very old realtek NIC ? ( They had a DMA issue on larger packets )
>
> If [...]

Hello! Thank you for your reply.
I check the image before the image converter. What I added was a to check for `grabResult.PayloadSize>0 and grabResult.PixelType!=-1` too and that helped. but I wanted to understand the `grabResult.GrabSucceeded()` method better.

```
# conecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
print("Using device ", camera.GetDeviceInfo().GetFullName())
# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
converter = pylon.ImageFormatConverter()
# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    if grabResult.GrabSucceeded() and grabResult.PayloadSize>0 and grabResult.PixelType!=-1:
        # Access the image data
        image = converter.Convert(grabResult)
        frame = image.GetArray()
        custom_method(frame)
        key = cv2.waitKey(1)
        if key == ( ord("q") or ord("Q") ):
            break
    else:
        print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
    grabResult.Release()
# Releasing the resource    
camera.StopGrabbing()
cv2.destroyAllWindows()
```

Packet size is 1500, and interestingly, I ran 5minutes test with Pylon's Bandwidth manager, and got full FPS speed without errors.

[image]

thiesmoeller, 2023-01-09:
The Bandwidth manager is internally also using `GrabSucceeded` .. this explains, why you see no error here.

Next step is to check if you are running on an older firmware, for which an update exists.
What is your model and fw version?

IgnacioEscorz, 2023-01-10:
Hello Thies,
thank ypu for your follow up.
Model is: acA5472-5gm
FW:  107769-11;U;acA5472_5g;V1.5-4;0