basler/pypylon #576: Silent crash when passing a reference to class function to pylon.ConfigurationEventHandler

TheRealRolandDeschain, 2023-01-04:
I have the following event handler:

```
class CamerRemovedEvent(pylon.ConfigurationEventHandler):
    """Event class to catch if a camera was disconnected during operation."""

    def __init__(self, callback: Callable):
        """Setting up of the event class.

        Args:
            callback (Callable): Reference of the method that should be called in case
            the event is triggered
        """
        self.callback = callback
        super().__init__()

    def OnCameraDeviceRemoved(self, camera: pylon.InstantCamera) -> None:
        """
        This method is called when a camera device removal from the
        PC has been detected.
        """
        self.callback()
```

I'm using this in a hardware-layer class, which basically serves as wrapper around the basler camera interface. This class has configuration and is responsible to handle everything with respect to the actual basler camera. After connecting I'm registering the Handler as follows:

```
# Register Event handle to detect camera removal
camera.RegisterConfiguration(
    CamerRemovedEvent(self._try_reconnecting),
    pylon.RegistrationMode_Append,
    pylon.Cleanup_Delete,
)
```

`self._try_reconnecting` basically just attempts to reconnect to the camera with the settings stored in the wrapper class (name, serial number, parameter,...) for x amount of times while waiting y amount of seconds between attempts. All of this works fine, but only if the camera was grabbing images (= `StartGrabbing()` was called and not yet stopped) during disconnection. However, if I only start the application and wait until everything is initialized (camera connected, parameters set,...) and then immediately disconnect the camera the following happens:

1. The debugger disconnects -> no breakpoints are hit anymore
2. The callback is called and correctly executed: I can see this in the logging output of the `self._try_reconnect()` function. If I reconnect the camera within the possible reconnection attempts, the function is also able to connect to the camera properly. 
3. Regardless of whether the camera was reconnected or not, the application closes without any errors. 

During testing I found that the issue does not appear, if we pass an external function. For example: 

```
# this is outside the class scope
def temp_function():
    print("testing")

# Register Event handle to detect camera removal
camera.RegisterConfiguration(
    CamerRemovedEvent(temp_function),
    pylon.RegistrationMode_Append,
    pylon.Cleanup_Delete,
)
```
will execute the the function as expected. 

I could just move the `_try_reconnecting()` function outside of the class scope, but in my particular setup it seems a bit hacky. Also
- I would like to understand what is happening here.
- Why does it work if the camera is active/grabbing images
- Is it a bug, or an error on my side?

TheRealRolandDeschain, 2023-01-04:
Additional Info:
- The whole application is wrapped into an FastAPI application (in case it is some uvicorn worker issue)
- The camera wrapper class is persistent even if the basler camera disconnects, so it is impossible that the reference to the class function is not available anymore.

TheRealRolandDeschain, 2023-01-05:
Some more testing:
- seems like the debugger is not disconnected, but I can't break on code within the `CamerRemovedEvent` class. If I remove the contents of `_try_reconnecting()` with a single print statement, it is executed normally and the application continues and can be debugged (breakpoints in the rest of the application are hit). However, a breakpoint at the print statement will not be hit.
- The silent crash seems  appears as soon as I try to write to `self.cam` during the `OnCameraDeviceRemoved` callback. `self.cam` is the member of my wrapper class, which holds the instance of the `pylon.InstantCamera`. Using `self.cam.Attach()` instead works fine.

thiesmoeller, 2023-01-05:
On device removal the camera object gets destructed.
I would recommend to instantiate a new instance.

This is how I handle these cases

TheRealRolandDeschain, 2023-01-05:
Hi, thanks for the reply!

_"I would recommend to instantiate a new instance."_
Currently to solve my issue above I do the following: 

> In `_try_reconnecting()` periodically call this function:
```
    def _connect_camera(self, serial_number: str) -> None:
        info = pylon.DeviceInfo()
        info.SetSerialNumber(serial_number)

        if self.cam is None:
            self.cam = pylon.InstantCamera(
                pylon.TlFactory.GetInstance().CreateDevice(info)
            )

            # Register Event handle to detect camera removal
            self.cam.RegisterConfiguration(
                CamerRemovedEvent(self._try_reconnecting),
                pylon.RegistrationMode_Append,
                pylon.Cleanup_Delete,
            )
        else:
            self.cam.DestroyDevice()
            self.cam.Attach(pylon.TlFactory.GetInstance().CreateDevice(info))
        logging.info(f"Established camera connection: {self.name}")
```
After that check if there is a connection (with `self.cam.IsPylonDeviceAttached()`) and if so, set all the necessary camera parameters and return. Both `_try_reconnect()` and `_connect_camera()` are functions of my wrapper class. 

This seems to work for me. If I try to completely destroy and reinstantiate `self.cam` during the CameraRemoved Event, I get the silent crash as described above. I have no idea what is causing the crash.

thiesmoeller, 2023-01-05:
Reason for the crash is most probably the "dark side" of pypylon where the python garbage collector and pylon's own removal of callback handlers are not properly synchronized yet :-/

TheRealRolandDeschain, 2023-01-05:
Thanks that makes things clearer for me :)