basler/pypylon #577: Only Two Channels in Basler acA1920-150uc Image Array, How to Display?

barmanroys, 2023-01-12:
Using this minimal code snippet (in a jupyter notebook) to capture and display an image from my Baseler acA1920. 

```python3
#!/usr/bin/env python3
# encoding: utf-8

from pypylon import pylon
import cv2
tl_factory = pylon.TlFactory.GetInstance()
devices = tl_factory.EnumerateDevices()
camera = pylon.InstantCamera()
camera.Attach(tl_factory.CreateFirstDevice())
camera.Open()
camera.StartGrabbing(1)
grab = camera.RetrieveResult(2000, pylon.TimeoutHandling_Return)
if grab.GrabSucceeded():
    img = grab.GetArray()
    print(f'Size of image: {img.shape}')
camera.Close()
print('The camera model is', devices[0].GetFriendlyName())
cv2.namedWindow('title', cv2.WINDOW_NORMAL)
cv2.imshow('title', img)
```

The img array size is (1200, 1920, 2). The first two dimensions match the documented resolution, but I wonder why are there only two channels instead of three. When I want to display it using CV2, this is the error I get. 

> Invalid number of channels in input image:
>     'VScn::contains(scn)'
> where
>     'scn' is 2

If it is important, my technology stack is as follows. 

- pypylon 1.8.0
- cv2 4.5.4
- python 3.10.6
- Ubuntu 22.04 
- Installed pylon and viewer from [the Baseler site](https://www.baslerweb.com/en/downloads/software-downloads/software-pylon-7-2-1-linux-x86-64bit-debian/
)

So what's the correct way to display the image in a jupyter notebook, and also pass it in response to an HTTP call if I am using fastapi and uvicorn?

thiesmoeller, 2023-01-12:
What you see is YUV422
 first channel is luminance Y
second  channel is chrominance UV interleaved.
This is best balance between image quality and bandwidth.

In your script you are not setting the PixelFormat. so the camera uses whatever is currently active ...
Depending on your camera model you can choose YUV
or also directly select BGR or RGB as PixelFormat.

To display a single image in jupyter I normally use matplotlib.pyplot.imshow as this creates inline images.

Regarding overall handling of jupyter and pypylon see our companion repository https://github.com/basler/pypylon-samples

barmanroys, 2023-01-13:
@thiesmoeller thanks a lot for the informative response, and filling a gap in my knowledge, as I was unaware of anything other than RGB to describe images.

closed 2023-01-13 by barmanroys