basler/pypylon #578: Low FPS on Jetson Nano with Basler daA4200-30mci

dmnaac, 2023-01-20:
As shown in the picture, the Basler daA4200-30mci camera has only 11 FPS at full resolution on Jetson Nano.
In Pylon Viewer, what features should be set to get a greater FPS at the resolution of 4192 x 3120?

[image]

MattsProjects, 2023-01-20:
Hi there, this is due to a bandwidth limitation on the host + using YUV422 format. When using YUV422, 11.5fps will be the limit if the host SoM + carrier board can only offer 2 mipi lanes to the host. If you can find one that provides 4 lanes, roughly 26fps should be possible. If you then switch to the lower bandwidth YUV420 format, the full 30fps is then possible.

dmnaac, 2023-01-21:
> Hi there, this is due to a bandwidth limitation on the host + using YUV422 format. When using YUV422, 11.5fps will be the limit if the host SoM + carrier board can only offer 2 mipi lanes to the [...]

Got it. Thanks a lot. :)

SMA2016a, 2023-01-22:
Check with your local distributor  to get a CEP package that supports  YUV420. You can get at least 15fps in your setup instead of 11.