basler/pypylon #579: Do you plan any support for ARM?

mouzkolit, 2023-01-20:
Currently there is no way to install the software on an M1/M2

thiesmoeller, 2023-02-02:
Apple silicon Support is planned. No release date yet.

Interim solution is to run in Rosetta emulated environment

brandonrwin, 2023-05-26:
At this time, the Mac Pro is the only remaining Intel computer that can be bought, from Apple. Choosing to use a Basler camera in your projects means you'll lose ~30% performance in your app, from Rosetta emulation.

thiesmoeller, 2023-05-27:
We have a first version running.
 Expect a first public release of pypylon for Apple Silicon  in the next weeks.

thiesmoeller, 2023-06-14:
Release 2.2.0 with full support of macOS x86_64 and arm64 ( >= macOS 11 ) is out.

Available via pip install or releases page.

closed 2023-06-14 by thiesmoeller