basler/pypylon #581: Can we use Basler cam on WSL platform?

nicolexi, 2023-01-27:
[image]
Hi I tried tp attached the Basler camera to the WSL platform but it failed and saying: 
usbip: error: Attach Request for 1-23 failed - Device in error state

usbipd: error: Failed to attach device with busid '1-23'.

thiesmoeller, 2023-02-02:
Never tried to get this running ... 
But as WSL is using USP/IP to transfer the data into the Linux domain ... This is might not be efficient enough to operate a high performance machine vision camera

thiesmoeller, 2023-02-02:
From your list of devices ... There is no Basler camera visible.

SMA2016a, 2023-02-08:
yes, I have tested only a GigE camera and it is working.

SMA2016a, 2023-02-09:
did you install pylon SDK on your Windows machine?  

[image]

SMA2016a, 2023-02-10:
I can confirm that USB camera is working in WSL2

jwongpic, 2023-04-04:
`PS C:\Windows\system32> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
2-4    1050:0407  USB Input Device, Microsoft Usbccid Smartcard Reader (WUDF)   Not attached
2-11   0c45:6720  Integrated Webcam                                             Not attached
2-14   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not attached
2-20   2676:ba02  Basler ace USB3 Vision Camera                                 Not attached

PS C:\Windows\system32> usbipd wsl attach --busid 2-20
usbipd: warning: Unknown USB filter 'CsDeviceControl' may be incompatible with this software; 'bind --force' may be required.
usbipd: warning: Unknown USB filter 'fuhub' may be incompatible with this software; 'bind --force' may be required.
usbip: error: recv op_common
usbip: error: query
usbipd: error: Failed to attach device with busid '2-20'.
`

I run into the same problem, I have the SDK installed. @SMA2016a , what are the commands you use to attach the camera to WSL?

SMA2016a, 2023-07-20:
yes, you need usb driver  also in WSL

closed 2024-05-31 by SMA2016a