basler/pypylon #582: The Camera Object Does not Follow the RAII Principle

barmanroys, 2023-02-03:
It is an amazingly helpful library, but I noticed the ```pypylon.pylon.InstantCamera``` class does not follow the RAII principle to open and close a camera. Instead, it leaves the acquisition (and eventual release) of the camera to developer judgement. I am curious if the maintainers had their reason for making this choice, and what possible scenarios are aided by this. 

Nonetheless, I thought some developers can benefit from this class that I developed as a context manager wrapper around the existing InstantCamera, that can replace the parent class respecting the Liskov substitution principle and can be used within a ```with``` block. 

```python3 
class CameraContext(py.InstantCamera):
    """This class is to respect the RAII principle of managing the camera resource."""

    def __enter__(self):
        """Acquire the camera."""
        self.Open()

    def __exit__(self, exc_type, exc_val, exc_tb):
        """Release the camera."""
        self.Close()
```

I can also add the methods to the existing class, if the maintainers are accepting contributions from the community. Kindly let me know the contributing guidelines for that.

thiesmoeller, 2023-02-06:
Thanks for your feedback.

You can always create a pull request to our project.

Looking at your proposal, I thought about embedding this context manager into the InstantCamera class itself ( which is already extended in a lot of places compared to the C++ functionality )

Only tricky thing might be, to check if this covers all paths to create/attach an instant camera

RoccoMatano, 2023-04-14:
>Only tricky thing might be, to check if this covers all paths to create/attach an instant camera

One thing to keep in mind when thinking about this: There are rare cases where you need an intermediate step between creating an `InstantCamera` and opening it. In my - admittedly quite unusual - work, I occasionally have to register a different configuration than the default (using RegisterConfiguration).