basler/pypylon #586: Noise Causes Trigger Problems for acA2500-60uc Camera

ridvanozdemir, 2023-02-20:
I am using acA2500-60uc - Basler ace model camera and  a sensor's output signal as a trigger. After a trigger recieved I don't need new one in next 1000 ms. but sometimes cause of electronic noises camera get trigger several times in seconds. I use "LineDebouncer" function but it's not enough.
How can I fix this problem? Can I add a wait command after grabbing ?

PolymaTh-EE98, 2023-02-20:
Try also falling edge on the trigger. Noise usually needs to be prevented though. Separate your cables for example.

thiesmoeller, 2023-02-27:
In addition to proper electrical setup, you can set the AcquisitionRate feature. This will set the minimum time between accepted trigger signals

closed 2023-02-28 by ridvanozdemir