basler/pypylon #587: PylonViewer working but not pypylon script Basler acA2000-50gc (NVIDIA Jetson Orin)

guillermoecn, 2023-02-23:
Hello, im having a trouble running scripts that grabs frames from my Basler camera acA2000-50gc.
The weird thing is that im able to run pylonviewer program and grab frames perfectly.
I've tried using pylon 6.2 and 7.2, and different versions of pypylon such as 1.7.0, 1.8.0 and 1.9.0 but nothing seems to work.
I've tried the same exact frame in my windows computer and works well.
Im using NVIDIA Jetson Orin, and i've tried the same scripts on another Jetson versions, woking well.

```
from pypylon import pylon
from pypylon import genicam

import sys

# Number of images to be grabbed.
countOfImagesToGrab = 100

# The exit code of the sample application.
exitCode = 0

try:
    # Create an instant camera object with the camera device found first.
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    camera.Open()

    # Print the model name of the camera.
    print("Using device ", camera.GetDeviceInfo().GetModelName())

    # demonstrate some feature access
    new_width = camera.Width.GetValue() - camera.Width.GetInc()
    if new_width >= camera.Width.GetMin():
        camera.Width.SetValue(new_width)

    # The parameter MaxNumBuffer can be used to control the count of buffers
    # allocated for grabbing. The default value of this parameter is 10.
    camera.MaxNumBuffer = 5

    # Start the grabbing of c_countOfImagesToGrab images.
    # The camera device is parameterized with a default configuration which
    # sets up free-running continuous acquisition.
    camera.StartGrabbingMax(countOfImagesToGrab)

    # Camera.StopGrabbing() is called automatically by the RetrieveResult() method
    # when c_countOfImagesToGrab images have been retrieved.
    while camera.IsGrabbing():
        # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
        grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        # Image grabbed successfully?
        if grabResult.GrabSucceeded():
            # Access the image data.
            print("SizeX: ", grabResult.Width)
            print("SizeY: ", grabResult.Height)
            img = grabResult.Array
            print("Gray value of first pixel: ", img[0, 0])
        else:
            print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
        grabResult.Release()
    camera.Close()

except genicam.GenericException as e:
    # Error handling.
    print("An exception occurred.")
    print(e.GetDescription())
    exitCode = 1

sys.exit(exitCode)
```
I get the next error:
```
An exception occurred. No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)
```

SMA2016a, 2023-02-23:
1) what is the IP and Subnet mask of the camera?
2) what is the subnet of NIC?

guillermoecn, 2023-02-23:
> 1. what is the IP and Subnet mask of the camera?
> 2. what is the subnet of NIC?

Hello, thank you for your quick response, i think i've solved it changing my ip from:
- Camera: 192.168.101.113 (netmask 255.255.255.0) to 192.168.200.113
- Jetson:   192.168.101.114 (netmask 255.255.255.0) to 192.168.200.114
Since my wifi (wlan0 interface) was 192.168.101.127 (netmask 255.255.254.0)
And if i take down the wlan0 interface the scripts works perfectly.
I think there might be some interference between the interfaces.

SMA2016a, 2023-02-24:
Network Interface Card (NIC)

OK this is just a routing issue.  Only one NIC should be in a single network.

guillermoecn, 2023-02-24:
> Network Interface Card (NIC)
> 
> OK this is just a routing issue. Only one NIC should be in a single network.

I appreciate your help!!!

closed 2023-02-24 by guillermoecn