basler/pypylon #591: Retrieving depth map from blaze 101

SonyPony, 2023-03-02:
Hello,

how can I retrieve depth map from the blaze 101 camera using pypylon API. I've checked the all the examples, but I couldn't find any information regarding the depth map.

SMA2016a, 2023-03-05:
Simply split the buffer in 2 halves.  First half would be the intensive image and the 3ed half  depth map.

SonyPony, 2023-03-06:
So if I understand it, I need to retrieve buffer data in byte format and then parse the "first" part as uint8 and the second part of the buffer as uint16. Where the second part represents the depth map. Am i right?

Shouldn't the the depth map contain more bytes than the intensity image, since it's in uint16?

Although, parsing bytes manually is a solution, I was hoping for a higher level API.

thiesmoeller, 2023-03-06:
@SonyPony .. the same happens in the GetArray method, that you use for 2d cameras.

Take a look at https://github.com/basler/pypylon/blob/c2a78fffffe2beab4d072c9a71bf53fac2a9c034/src/pylon/GrabResultPtr.i#L61

With GetBuffer you get the raw data and the ndarray constructor will create a numpy array for you.

SonyPony, 2023-03-06:
@thiesmoeller  @SMA2016a thank you for your time and answers. That answered my question.

closed 2023-03-06 by SonyPony

cknox-delsys, 2024-05-09:
Hi! I'm trying to retrieve both the intensity image _and_ the depth image, but your suggestion to use the first half of the buffer does not seem to work for me. My buffer size is always 606720 which fits a uint16 (640,480) image. I've only been able to change the buffer size if I switch to `Coord3D_ABC32f` for my pixel format which gives a buffer size of 3640320. However when I run the blaze viewer application (which displays both the intensity and depth image together) I see that the buffer size is 4853760. I've tried just about every configuration of the ComponentSelector and PixelFormat without any luck. Do you have any pointers?

Here's my code:
```
from pypylon import pylon
import cv2
import numpy as np

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        buffer = grabResult.GetImageBuffer()
        buf_len = grabResult.GetPayloadSize()

        num = grabResult.GetDataComponentCount()
        print(f"Num Components: {num}")
        print(f"Buf Len : {buf_len}")

        shape, dtype, fmt = grabResult.GetImageFormat()
        depth_img = np.frombuffer(buffer[:buf_len], dtype=dtype).reshape(shape)
```

AndyPermaRobotics, 2024-07-01:
Hi @cknox-delsys! 
How did you use the ComponentSelector in pypylon?
Can you provide a short code snippet?

thiesmoeller, 2024-07-01:
If you have short time to wait ...
The upcoming pypylon release has full support for Blaze 3D data.