basler/pypylon #593: Synchronous operation of multiple cameras (acA2500-14gm)

TuanHAnhVN, 2023-03-18:
Hi everyone,

My project is using 5 cameras (Basler acA2500-14gm) to check/inspect objects. But I have meet a problem that related to read images from 5 cameras: The time of reading image is so much slow (in synchronicity). and sometimes, cannot capture image. How to improve it?

Current status:
I have 5 sub-process to capture from each camera (each camera is run on a sub-process).
I use a switch that supports PoE but not support PTP. 

Please help me! Thank you so much.

thiesmoeller, 2023-03-18:
Hi @TuanHAnhVN,

What is the required synchronicity in microseconds of the five cameras that you have to reach ( assuming from your description that they should take their images at the "same time")?

What is the target framerate of your cameras and the Pixelformat?

TuanHAnhVN, 2023-03-19:
> Hi @TuanHAnhVN,
> 
> What is the required synchronicity in microseconds of the five cameras that you have to reach ( assuming from your description that they should take their images at the "same time")?
> 
> What is the target framerate of your cameras and the Pixelformat?

Hi @thiesmoeller ,
Yes, as you said, i want to take images at the same time. It means the less time the better. I hope I can reach 10fps ~ 100ms for each image. But I could not, sometimes when I take the image I get error like pixelformat or len array=0. It takes me more time to take it again. That makes it long to take the frame from 5 cameras in same time.
 I'm using 5 camera acA2500-14gm and a switch (**not support PTP**).

TuanHAnhVN, 2023-03-19:
After reading more documents, I found that my kind of camera - acA2500-14gm - is not support PTP and 5GigE. So it could not take image faster than 14fps. But I still find the best solution to solve my case. If you have any idea please support me. I am so grateful for that!

thiesmoeller, 2023-03-19:
If a camera model has no PTP support you have to use classic physical trigger line with a 10hz signal.

As a side note you can run PTP cameras on a no -PTP switch ... The precision is lower ( 10s of microseconds instead of nanoseconds) but depending on your requirements this might be sufficient)

To the question if this will work at all:
Has your switch a 5/10Gbit uplink  to your PC?

As (2592*1944)*10_fps*5_cams are ~240MB/s  you need more than a standard 1Gbit switch to simply transfer the data to your host in time.

TuanHAnhVN, 2023-03-19:
> If a camera model has no PTP support you have to use classic physical trigger line with a 10hz signal.
> 
> As a side note you can run PTP cameras on a no -PTP switch ... The precision is lower ( 10s of microseconds instead of nanoseconds) but depending on your requirements this might be sufficient)

My camera is acA2500-14gm. So it doesnt support PTP.
> 
> To the question if this will work at all: Has your switch a 5/10Gbit uplink to your PC?
> 
> As (2592*1944)_10_fps_5_cams are ~240MB/s you need more than a standard 1Gbit switch to simply transfer the data to your host in time.

Currently, my switch support 1GB. Is it ok?

TuanHAnhVN, 2023-03-19:
> If a camera model has no PTP support you have to use classic physical trigger line with a 10hz signal.
>
> As a side note you can run PTP cameras on a no -PTP switch ... The precision is lower ( 10s of [...]

Hi @thiesmoeller if you have any idea for my case/my system, please help me. I can not change switch or kind of camera. :(. I want to take the image in the best way at the same time. Thank you so much!

thiesmoeller, 2023-03-19:
The best way to capture in sync is via HW trigger of the cameras. 

What is the model of your switch ? 
Maybe it has Link Aggregation feature that would allow to connect to multiple network cards in your PC in parallel.

If not you are limited by the bandwidth of 1Gbit to your PC.
This means ~100MB/s peak.
So you if you need 10fps from 10 cameras you would have  use either RegionOfInterest or Binning to reduce the amount of data sent by each camera

peci1, 2024-06-27:
Just for reference, Basler cameras do not require any PTP support in switches. They can be configured (and are by default) to use the E2E delay mechanism, which is dumb-switch-friendly.

Of course, if you have a switch that can fill the switching delay in the packets, you'll get better sync precision/lower jitter.