basler/pypylon #594: How to obtain 12 bit pixel-depth images from Basler camera?

laser-vis, 2023-03-21:
I have a puA2500-14uc as well as a puA2500-14um, both capable of 12-bit pixel depth. I followed the discussion in [ issue #38](https://github.com/basler/pypylon/issues/38) to obtain the maximum possible bit depth from the color camera, which provided a dramatic improvement in dynamic range in my images. However, when I use similar code for the monochrome camera, the Mono8 and Mono12 images look identical even though the latter is a "16-bit" image (only the 12 LSBs are meaningful).

This is how I am setting up the camera (puA2500-14um):

```
converter = pylon.ImageFormatConverter()
camera.PixelFormat.SetValue("Mono12")
converter.OutputPixelFormat = pylon.PixelType_Mono16
converter.OutputBitAlignment = pylon.OutputBitAlignment_LsbAligned
```
I read the image simply by using:
```
image = converter.Convert(grabResult)
img = image.GetArray()
```
The output image does _appear_ to vary between 0-4095, but there is no visual difference between the 8-bit and 12-bit images. I do understand that in order to display the 12-bit image using `cv.imshow()` I need to scale it by a factor of 16.

Am I missing something?

thiesmoeller, 2023-03-21:
From the interface side, you could always check on missing codes like below code ... or with an histogram.
The Mono8 output of basler cameras is typically the MSB 8bit of image sensor data. In Mono12 all data is output.

```python
import pypylon.pylon as py
import matplotlib.pyplot as plt
import numpy as np
```

```python
tlf = py.TlFactory.GetInstance()
cam = py.InstantCamera(tlf.CreateFirstDevice())
```

```python
cam.Open()
```

```python
# capture in Mono8
cam.PixelFormat.SetValue("Mono8")
img_8 = cam.GrabOne(1000).GetArray()
print("8bit: ", bin(0xff))
print("img:  ", bin(np.bitwise_or.reduce(img_8, axis=(0,1))))

```

    8bit:  0b11111111
    img:   0b11111111

```python
# simulate your described "shifted" mono8
cam.PixelFormat.SetValue("Mono8")
img_8 = cam.GrabOne(1000).GetArray().astype(np.uint16)
img_12_shifted = img_8 << 4
print("12bit: ", bin(0xfff))
print("img:   ", bin(np.bitwise_or.reduce(img_12_shifted, axis=(0,1))))
```

    12bit:  0b111111111111
    img:    0b111111110000

```python
# capture in Mono12
cam.PixelFormat.SetValue("Mono12")
img_12 = cam.GrabOne(1000).GetArray()
print("12bit: ", bin(0xfff))
print("img:   ", bin(np.bitwise_or.reduce(img_12, axis=(0,1))))
```

    12bit:  0b111111111111
    img:    0b111111111111

laser-vis, 2023-03-21:
Once again, @thiesmoeller thanks for the detailed response. I was already starting to look at Hisograms, and your detailed analysis completes my education on this topic.

My expectations were set incorrectly by the difference in image quality between the "YCbCr422_8" and "BayerGB12" pixel formats - I mistakenly attributed the difference to 8 vs. 12 bits.

closed 2023-03-25 by laser-vis