basler/pypylon #595: PTP Configuration (acA5472-5gm)

Magic-Doctor, 2023-03-22:
Hi,  

We adopt the "**Scheduled Action Command**" feature because we need to achieve microsecond-level multi-camera synchronization. I learned from the documentation that setting the master clock and slave clock is necessary when configuring PTP clock synchronization, but it is not explicitly stated how to invoke the API to set this for the acA5472-5gm camera model（**Priority 1** is not supported）. Additionally, I also found that it is not possible to modify the "IEEE 1588 Status (Latched)" status in Pylon Viewer software (all cameras are default masters). Is there anything I didn't notice? Could you give some suggestions on how to address this issue? Is there an API available for this purpose? Any assistance provided would be greatly appreciated!

thiesmoeller, 2023-03-24:
What is the network architecture between your cameras and your host?

Magic-Doctor, 2023-03-24:
I conducted tests using two cameras. One camera was connected to the computer's network port via a Gige cable, while the other camera's Gige cable was connected to a network adapter, which was then connected to the computer via USB. I think I understand a bit now, perhaps using a network switch could solve this problem. Is that correct?

thiesmoeller, 2023-03-24:
Yes this is correct.
Both cameras have to either 

* Share the same Ethernet
* You run a PTP daemon on your host that synchronizes the two separate ethernets ( on Linux ptp4l )