basler/pypylon #597: The buffer was incompletely grabbed (1.9.0 and 1.8.0) but working on 1.7.4

daglofthus, 2023-03-23:
Camera: Basler a2A5328-4gmBAS
Hardware: Linux 5.15.0-67-generic #74~20.04.1-Ubuntu SMP Wed Feb 22 14:52:34 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
Python version: 3.8
MTU sat to max.

When grabbing frames with pypylon==1.7.4 I receive frames as expected. Once I try to upgrade to either 1.8.0 or 1.9.0 I get the incompletely grabbed error:
`The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size.`

SMA2016a, 2023-04-02:
did you also set the packet size to a high value like 8000?

Also set the interpacket delay to a higher value. Simply try in 1000 ticks steps until the error is not occuring.

daglofthus, 2023-04-14:
This is what I set in code today:
```
self.device.GevSCPSPacketSize.SetValue(self.device.GevSCPSPacketSize.GetMax())
sbs_node = self.device.StreamGrabber.GetNode('SocketBufferSize')
sbs_node.SetValue(sbs_node.GetMax())
```
I tried your suggestion on increasing interpacket delay. But even with the max possible `GevSCPD.GetMax()` I was not able to retrieve the frames. For reference, I have no issues retrieving frames with interpackage delay at 512 with pypylon==1.7.4

daglofthus, 2023-04-17:
I got the newest firmware for the camera from Basler, and it seemed to do the trick. Closing

closed 2023-04-17 by daglofthus