basler/pypylon #598: Ensure synchronized grabbing from multiple cameras

dahleri, 2023-04-03:
Hi,
I am building a stereo application where it is important to ensure that the processed frames were acquired synchronous at the same time, in order to triangulate points from the different views.  I am syncing the image acquisition through PTP and periodic signal. The computing time for the stereo algorithm is very unstable and sometimes it takes longer than the fixed framerate. I struggle to decide on the most appropriate grabbing strategy. 

I would like to use "LatestImageOnly" to be able to compare the output from my stereo application to other sensors, but for some reason it sometimes skips frames from one or more of the cameras. I dont know why this happens but could maybe be because the images are not transferred at the same time and the code therefore skip one of the frames if its not ready in the memory? Is there a way to get around this problem?

I must keep control of when the image was acquired in order to compare my output to other sensors, therefore I dont want a queue building up. At the same time, I must ensure that the images were acquired at the same time and cannot allow one camera to skip an image and loose sync. 

How would you approach this problem? Thank you very much in advance!

SMA2016a, 2023-04-04:
[I would like to use "LatestImageOnly" to be able to compare the output from my stereo application to other sensors, but for some reason it sometimes skips frames from one or more of the cameras]  Most properly, the image processing takes long time than 1/fps
use onebyone instead of LatestImageOnly

set Packetsize to 8000
interpacketdelay  xxxx   test a value by increasing it by 500 steps.

dahleri, 2023-04-05:
Thank you for your response! To clarify, its not a problem that we sometimes skip images. This is rather expected since I know the processing sometime takes longer than 1/fps. The problem is when only one camera skips an image, because we then lose sync which can be seen comparing the timestamps. 

I am confused about how this happens. I would ideally always retrieve the latest acquired images from both cameras and be able to trust that they both were taken at the same time. 

I will try with the packetsize and interpacketdelay, but do you think they will have any impact if the problem is not buffer underrun?

thiesmoeller, 2023-04-05:
From system architecture point of view I would recommend to decouple the stereo processing and the capturing of the image pairs.

The pylon APIs will only take care of the images per camera. But your application has to handle "LatestPairOnly"...

E.g. you could e.g. capture the pairs and push them into a fixed length queue. 
And run your stereo code on the other end of the queue.

dahleri, 2023-04-06:
Do you mean that the acquisition and processing are done in separate threads and buffers stored in a customised buffer factory? 

Will have a look at the samples Grab_UsingBufferFactory and Grab_UsingGrabLoopThread to test if that solves my problem. Will let you know the results as soon as I have managed to implement the suggestions.

thiesmoeller, 2023-04-06:
You don't have to use buffer factory.
Just have one grab thread that handles the cameras and provide pairs of images in a queue 
Another thread then does the stereo processing and reads from this queue.

thiesmoeller, 2023-04-12:
Hi @dahleri ,

the issue with your setup is, that you enforce an order of the receive of the data from the two cameras.
This can might lead to your problem. You wait for buffer-left while buffer-right is already overwritten by 'LatestImageOnly' policy.

It is recommended in these setups to use the Background loop pattern as described in https://github.com/basler/pypylon-samples/blob/main/notebooks/grabstrategies.ipynb or in https://github.com/basler/pypylon/blob/master/samples/grabusinggrabloopthread.py

You can attach the same OnImageGrabbed handler to both cameras. So you have a single place to sort the incoming buffers.
On each camera you can set a CameraContext, that gets reflected in the GrabResult.