basler/pypylon #603: Receiving error when using Data Chunk

AIGR-sw, 2023-04-18:
I would like to operate four GigE cameras (a2A2448-23gcBAS) synchronously using Baslers 4 port PoE GigE interface card.
For this purpose, I use of the _Precision Time Protocol (PTP)_ together with the _Periodic Signal_ feature to capture images synchronously on all cameras at regular time intervals.

Once the images are received, I need to sort them so that images taken at the same time are grouped together (they do not arrive in the correct order).

To know, which images were taken together I add a counter value to each image as _Data Chunk_.
This counter value is reset on all cameras at the same time using the _Action Command_ feature.
The counter is incremented every time an image is triggered by the _Periodic Signal_.

Up to this point, everything works well.
However, from time to time it happens that the data of a camera cannot be received.
The program aborts at the line ```cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:``` with the following error:

```bash
Traceback (most recent call last):
  ...
    with self.cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
  File "/home/vscode/.local/lib/python3.8/site-packages/pypylon/pylon.py", line 4633, in RetrieveResult
    return _pylon.InstantCameraArray_RetrieveResult(self, *args)
_genicam.RuntimeException: Invalid layout of buffer attached to chunk parser! : RuntimeException thrown (file 'ChunkAdapterGEV.cpp', line 97)
```

This error only disappears when I restart the cameras.

Two questions:
- Is there a simpler, less error-prone way to get a synchronization identifier for each frame (as an alternative to _Data Chunks_ and _Action Command_).
- Where does the error described above come from? How can I get rid of it?

Below my Python implementation, using Pypylon 1.9.0 and the features mentioned above.

```python
from pypylon import pylon, genicam
import warnings

framerate = 27
action_key = 0x4711
group_key = 0x112233
group_mask = pylon.AllGroupMask

################################################################################
#
# Initialize GigE cameras
#
################################################################################

# Get the transport layer factory
tlFactory = pylon.TlFactory.GetInstance()

# Get all attached devices and exit application if no device is found
devices = tlFactory.EnumerateDevices()
cam_count = len(devices)
if not cam_count:
    raise EnvironmentError('No camera device found')

# Create and attach all Pylon Devices
cameras = pylon.InstantCameraArray(cam_count)
for i, (camera, device) in enumerate(zip(cameras, devices)):
    camera.Attach(tlFactory.CreateDevice(device))
    camera.SetCameraContext(i)
cameras.Open()

# Set default parameters
for i, camera in enumerate(cameras):
    camera.UserSetSelector.SetValue("Default")
    camera.UserSetLoad.Execute()

# Initialize PTP and check initialization
for i, camera in enumerate(cameras):
    _init_ptp(camera, i)

success = _check_ptp(cameras)
if not success:
    raise EnvironmentError('PTP initialization was not successful')

# Initialize general camera parameters
for i, camera in enumerate(cameras):
    # Transport Layer Control
    camera.GevSCPSPacketSize.SetValue(8192)
    camera.GevSCPD.SetValue(7643)
    camera.GevSCFTD.SetValue(0)
    camera.BandwidthReserveMode.SetValue("Performance")

    # Periodic Signal setup
    if camera.BslPeriodicSignalSource.GetValue() != 'PtpClock':
        warnings.warn('Clock source of periodic signal is not `PtpClock`')
    camera.BslPeriodicSignalPeriod.SetValue(1 / framerate * 1e6)
    camera.BslPeriodicSignalDelay.SetValue(0)
    camera.TriggerSelector.SetValue("FrameStart")
    camera.TriggerMode.SetValue("On")
    camera.TriggerSource.SetValue("PeriodicSignal1")

    # Prepare counter 1
    camera.CounterSelector.SetValue("Counter1")
    camera.CounterReset.Execute()
    camera.CounterEventSource.SetValue("PeriodicSignal1")
    camera.CounterResetSource.SetValue("Action1")

    # Enable data chunks
    camera.ChunkModeActive.SetValue(True)
    camera.ChunkSelector.SetValue("CounterValue")
    camera.ChunkCounterSelector.SetValue("Counter1")
    camera.ChunkEnable.SetValue(True)

    # Action command
    camera.ActionDeviceKey.SetValue(action_key)
    camera.ActionGroupKey.SetValue(group_key)
    camera.ActionGroupMask.SetValue(group_mask)

    # Trigger action command
    gige_tl = tlFactory.CreateTl('BaslerGigE')
    act_cmd = gige_tl.ScheduledActionCommand(
        deviceKey=action_key,
        groupKey=group_key,
        groupMask=group_mask,
        actionTimeNs=0,
        )
    act_cmd.IssueNoWait()

# Image decompression
descriptor = cameras[0].BslImageCompressionBCBDescriptor.GetAll()
decompressor = pylon.ImageDecompressor()
decompressor.SetCompressionDescriptor(descriptor)

# Prepare image grabbing
cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

################################################################################
#
# Capture images
#
################################################################################

# Prepare frame synchronization
imgs = [None] * cam_count
img_ids = [None] * cam_count

while True:
    with cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
        if grabResult.GrabSucceeded():
            decompressed_image = decompressor.DecompressImage(grabResult.Buffer)
            decompressed_array = decompressed_image.Array
            decompressed_image.Release()
            cam_idx = grabResult.GetCameraContext()
            frame_idx = grabResult.ChunkCounterValue.GetValue()

            imgs[cam_idx] = decompressed_array
            img_ids[cam_idx] = frame_idx

        # If set of images is available
        if img_ids.count(img_ids[0]) == len(img_ids):
            pass
            # Do some processing ...
        else:
            print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
```

SMA2016a, 2023-04-26:
you can put all grabresult into a list and search for all grabresult +-  5000ns.
The timestamp ticks is 1ns.

SMA2016a, 2023-04-27:
no,  since the clock of all cameras are synchronized by PTP, it will work

AIGR-sw, 2023-04-28:
Which timestamp do you mean? It does not work if I use the the timestamp which is attached to the grab result by default (```grabResult.TimeStamp``` in the following example).

The clock is synchronized, but the time stamp value is not identical if the four cameras are not started at the same time.

```python
while True:
    with self.cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult :
        if grabResult.GrabSucceeded():
            try:
                decompressed_image = self.decompressor.DecompressImage(grabResult.Buffer)
            except genicam.RuntimeException as e:
                warnings.warn(str(e))
                imgs = None
            else:
                decompressed_array = decompressed_image.Array
                decompressed_image.Release()
                cam_idx = grabResult.GetCameraContext()
                timestamp = grabResult.TimeStamp * 1e-9
                imgs = frame_buffer.group(timestamp, cam_idx, decompressed_array)
```
Is there any way I can reset this timestamp on all devices at the same time?

thiesmoeller, 2023-04-29:
What is your network setup?
To get synchronized cameras they have to be all in the same ethernet.
So either using a switch or a PTP daemon on your host that takes care to sync all NICs if you use more than one

AIGR-sw, 2023-04-29:
> What is your network setup? To get synchronized cameras they have to be all in the same ethernet. So either using a switch or a PTP daemon on your host that takes care to sync all NICs if you use more than one

As mentioned above the four cameras are all connected to a [4 Port PoE Basler GigE Interface Card](https://www.baslerweb.com/en/products/acquisition-cards/gige-interface-cards/basler-gige-interface-card-4-port-poe/) which should be capable of PTP according to the specification.

Most of the time it works fine. But from time to time the error occurs which I described in the first post:

```bash
Traceback (most recent call last):
  ...
    with self.cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
  File "/home/vscode/.local/lib/python3.8/site-packages/pypylon/pylon.py", line 4633, in RetrieveResult
    return _pylon.InstantCameraArray_RetrieveResult(self, *args)
_genicam.RuntimeException: Invalid layout of buffer attached to chunk parser! : RuntimeException thrown (file 'ChunkAdapterGEV.cpp', line 97)
```

This error only occurs if I use the _Data Chunks_ feature which I need to group the received images (see detailed description above).

thiesmoeller, 2023-04-29:
In this setup, which PTP daemon do you use ?

ptpd or ptp4l?

AIGR-sw, 2023-05-01:
To be honest, I don't know.

I did the installation according to [Basler's instructions](https://docs.baslerweb.com/software-installation-(gige-interface-cards)). An Ubuntu 20.04 is running on my PC. Framegrabbing and PTP worked on the first try, without me having to install any additional software.

How can I find out the type of PTP daemon?

SMA2016a, 2023-05-01:
<To get synchronized cameras they have to be all in the same ethernet>   meaning this, you need to connect all camera to a 8 port network switch  and connect 4 cables to the network switch to ensure needed bandwidth. 

Alternative, you  can try to enable bridge mode on linux.

AIGR-sw, 2023-05-01:
> meaning this, you need to connect all camera to a 8 port network switch and connect 4 cables to the network switch to ensure needed bandwidth.
> 
> Alternative, you can try to enable bridge mode on linux.

I think you don't quite understand my problem.

The setup with the [Basler 4-port GigE interface card](https://www.baslerweb.com/en/products/acquisition-cards/gige-interface-cards/basler-gige-interface-card-4-port-poe/) and the four [Basler cameras](https://www.baslerweb.com/en/products/cameras/area-scan-cameras/ace2/a2a2448-23gcpro/) works fine. I don't need a switch and bandwidth is not my problem. I checked it with the [Bandwidth Manager](https://docs.baslerweb.com/bandwidth-manager). The PTP also seems to work fine.

The problem is that occasionally, without apparent cause, the following error occurs:

```bash
Traceback (most recent call last):
  ...
    with self.cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
  File "/home/vscode/.local/lib/python3.8/site-packages/pypylon/pylon.py", line 4633, in RetrieveResult
    return _pylon.InstantCameraArray_RetrieveResult(self, *args)
_genicam.RuntimeException: Invalid layout of buffer attached to chunk parser! : RuntimeException thrown (file 'ChunkAdapterGEV.cpp', line 97)
```

thiesmoeller, 2023-05-01:
The precision time protocol relies on having all cameras in the same ethernet.

In your case you have four separate ethernets.
I assume that all  cameras signal being in the clock master role in your case.

Solution to your setup is:

Use an external switch to bridge the four connections into one ethernet.

Create a SW bridge between the four ports. This will lead to degraded PTP performance.

Install a PTP daemon e.g. ptp4l on your system. This will take care to synchronize the four ports.

SMA2016a, 2023-05-09:
install the software using the following command

apt-get install ptpd

start the software using following 

ptpd -c /etc/aceClassic_PTPSlave.conf

or

./ptp4l -i eno1 -m --priority1 1 -l 7 --logSyncInterval -1

you can also pass -i eno1 -i eno2 etc  multiple NIC names.

Please also refer this instruction, since this is a generic tool, you will find a lot of information in internet.
https://docs.fedoraproject.org/en-US/fedora/latest/system-administrators-guide/servers/Configuring_PTP_Using_ptp4l/

I do not know how to upload the aceClassic_PTPSlave.conf.  if you send me a private message I will send you a tested configuration.

SMA2016a, 2023-05-09:
; ========================================
; PTPDv2 version 2.3.2 default configuration
; ========================================

; NOTE: the following settings are affected by ptpengine:preset selection:
;           ptpengine:slave_only
;           clock:no_adjust
;           ptpengine:clock_class - allowed range and default value
; To see all preset settings, run ptpd2 -H (--long-help)

; Network interface to use - eth0, igb0 etc. (required).
ptpengine:interface = enp0s8

; Backup network interface to use - eth0, igb0 etc. When no GM available, 
; slave will keep alternating between primary and secondary until a GM is found.
; 
ptpengine:backup_interface = 

; PTP engine preset:
; none	     = Defaults, no clock class restrictions
; masteronly  = Master, passive when not best master (clock class 0..127)
; masterslave = Full IEEE 1588 implementation:
; Master, slave when not best master
; (clock class 128..254)
; slaveonly   = Slave only (clock class 255 only)
; 
; Options: none masteronly masterslave slaveonly 
ptpengine:preset = masteronly

; Transport type for PTP packets. Ethernet transport requires libpcap support.
; Options: ipv4 ethernet 
ptpengine:transport = ipv4

; Disable PTP port. Causes the PTP state machine to stay in PTP_DISABLED state indefinitely,
; until it is re-enabled via configuration change or ENABLE_PORT management message.
ptpengine:disabled = N

; IP transmission mode (requires IP transport) - hybrid mode uses
; multicast for sync and announce, and unicast for delay request and
; response; unicast mode uses unicast for all transmission.
; When unicast mode is selected, destination IP(s) may need to be configured
; (ptpengine:unicast_destinations).
; Options: multicast unicast hybrid 
ptpengine:ip_mode = multicast

; Enable unicast negotiation support using signaling messages
; 
ptpengine:unicast_negotiation = N

; When using unicast negotiation (slave), accept PTP messages from any master.
; By default, only messages from acceptable masters (ptpengine:unicast_destinations)
; are accepted, and only if transmission was granted by the master
; 
ptpengine:unicast_any_master = N

; PTP port number wildcard mask applied onto port identities when running
; unicast negotiation: allows multiple port identities to be accepted as one.
; This option can be used as a workaround where a node sends signaling messages and
; timing messages with different port identities
ptpengine:unicast_port_mask = 0

; Disable Best Master Clock Algorithm for unicast masters:
; Only effective for masteronly preset - all Announce messages
; will be ignored and clock will transition directly into MASTER state.
; 
ptpengine:disable_bmca = N

; When unicast negotiation enabled on a master clock, 
; reply to transmission requests also in LISTENING state.
ptpengine:unicast_negotiation_listening = N
[... 742 lines cut ...]
; 
global:timingdomain_election_delay = 15

; Enable NTPd integration
ntpengine:enabled = N

; Enable control over local NTPd daemon
ntpengine:control_enabled = N

; NTP control check interval in seconds
; 
ntpengine:check_interval = 15

; NTP key number - must be configured as a trusted control key in ntp.conf,
; and be non-zero for the ntpengine:control_enabled setting to take effect.
; 
ntpengine:key_id = 0

; NTP key (plain text, max. 20 characters) - must match the key configured in
; ntpd's keys file, and must be non-zero for the ntpengine:control_enabled
; setting to take effect.
; 
ntpengine:key = 

; ========= newline required in the end ==========

AIGR-sw, 2024-03-25:
@thiesmoeller @SMA2016a 
Still no solution for this issue?

SMA2016a, 2024-03-28:
error while using chunk is more or less performance issue  of the total hardware. Nothing released pypylon that we are support in this forum.  Please turn to your distributor channel or your local support team to get help   in further troubleshooting

AIGR-sw, 2024-03-28:
> error while using chunk is more or less performance issue of the total hardware. Nothing released pypylon that we are support in this forum. Please turn to your distributor channel or your local support team to get help in further troubleshooting

My question is related to the configuration of the PTP daemon for a 4-port Ethernet card as described in the last comment, not to the problem of data chunks.

Nailtha, 2024-04-17:
> > ./ptp4l -i eno1 -m --priority1 1 -l 7 --logSyncInterval -1
> 
> Unfortunately, the following command
> 
> ```shell
> sudo ptp4l -i enp4s0f0 -i enp4s0f1 -i enp4s0f2 -i enp4s0f3 -m --priority1 1 -l 7 --logSyncInterval -1
> ```
> 
> returns the following error message:
> 
> ```shell
> ptp4l[2010.473]: port 2: PHC device mismatch
> ptp4l[2010.473]: port 2: /dev/ptp0 requested, ptp1 attached
> ptp4l[2010.473]: failed to open port enp4s0f1
> failed to create a clock
> ```
> 
> Any idea why?

I can confirm I am facing the exact same issue using a Basler 4-port Ethernet Card and two Basler ACE-2 cameras.
Running the following command:

`$ sudo ptp4l -i eth0 -i eth1 -m`

results in the same error: 

```ptp4l[540554.048]: selected /dev/ptp0 as PTP clock
ptp4l[540554.049]: port 2 (eth1): PHC device mismatch
ptp4l[540554.049]: port 2 (eth1): /dev/ptp0 requested, ptp1 attached
ptp4l[540554.049]: failed to open port eth1
failed to create a clock
```

SMA2016a, 2024-04-20:
check if this works?

sudo ptp4l -i eth0 -i eth1 -s 

and would post the output of ethtool  -t

Nailtha, 2024-04-20:
> check if this works?
> 
> sudo ptp4l -i eth0 -i eth1 -s
> 
> and would post the output of ethtool -t

I don't want to hijack the thread but in case in also helps others, `sudo ptp4l -i eth0 -i eth1 -s` is not working (same error as above) and the output of ethtool -T is below:

```
Time stamping parameters for eth0:
Capabilities:
    hardware-transmit
    software-transmit
    hardware-receive
    software-receive
    software-system-clock
    hardware-raw-clock
PTP Hardware Clock: 0
Hardware Transmit Timestamp Modes:
    off
    on
Hardware Receive Filter Modes:
    none
    all
```

see also [here](https://serverfault.com/questions/1158237/ptp-and-ethtool-output-on-nvidia-jetson-orin) where I posted the ethtool output which seems a bit off.

Thanks