basler/pypylon #604: grabResult.GrabSucceeded() = False Frequently | 6 GigE Camera Sequential Capturing

pratiksparwatwar, 2023-04-28:
### Existing Setup
Server : HP Proliant dl380, 
Processor : Intel Xeon Silver 4210 CPU @ 2.40GHz x 40
Ram : 64 GB, HardDisk : 1 TB
Operating system : Ubuntu 20.04 desktop
pylon version : 7.2.1
pypylon version : 1.8.0
Camera : Basler Ace GigE

### Problem
I am capturing image from 6 Basler ace GigE camera in a sequential manner through single python code.
In my current application I need to kill and restart the capturing code few times in a day.
After every restart, For some cameras, grabresult.Grabsucceded is not getting True. 
But After restarting few times the code runs fine and images from all the cameras are captured. 

Even if the code is not restarted, 
Some of the cameras are not getting the grabresult for an hour and then automatically they start grabing again. This happens for 2-3 times in a day.
The issue is not related to any particular camera. It is 1-2 random cameras out of the 6.
After server reboot the code runs smoothly but after several restarts of the script the issue persist. 

The issue started a couple of days ago. The code ran smoothly for a 20-22 days. 

Also, In case I connect the cameras to my laptop, I am not getting any error on my laptop if I continuously kill the process and restart the frame capturing multiple times. Everytime the grabresult.Grabsucceded is True.
Instead of rack server if CPU server is used then the issue is not coming.

So I don't if the issue related the rack server or there are some settings needs to be done to use the cameras on rack server
 
```
import cv2
import time
import os
import threading
import logging
from pypylon import pylon
import traceback
import multiprocessing as mp
#for recording time limit
import queue
import shutil
import datetime
import pymysql
import signal
import subprocess

img_number=-1
async_q=queue.Queue()
grab_state=True
frame_ctr=0
mv_logger = None
frame_logger = None

csm1frontcam = "40271839"
csm1s1cam = "40271837"
csm1s2cam = "40271844"

csm2frontcam = "40267929"
csm2s1cam = "40271836"
csm2s2cam = "40271846"

exposure_csm1frontcam = 15000
exposure_csm1s1cam = 15000
exposure_csm1s2cam = 15000

exposure_csm2frontcam = 15000
exposure_csm2s1cam = 15000
exposure_csm2s2cam = 15000

fpsVar = 3.00

processID = os.getpid()
[... 497 lines cut ...]
        #mv_logger.debug("MV Record Process Started")
        self.init_cam()
        #mv_logger.debug("Process ended")      

def run_once(f):
    def wrapper(*args, **kwargs):
        if not wrapper.has_run:
            wrapper.has_run = True
            return f(*args, **kwargs)
    wrapper.has_run = False
    return wrapper

def call_error(cam_pos):
    print("Error in function my function "+str(cam_pos))

if __name__=="__main__":
    obj1=mvrecordingObj(os.getcwd(), "FNL_TST", 60)
    obj1.run_module()

```
Also multiple times, 
Even after grabResult.GrabSucceeded() is True, The following error comes up at conversion
`Cannot convert image. The passed source image is invalid. : InvalidArgumentException thrown (file 'ImageFormatConverter.cpp', line 77)
`
Please let me know what can be the problem and probable solution. Thank you.

thiesmoeller, 2023-04-29:
Your cameras are free running. You wrote your code in sequential order, but this doesn't mean, that the images are strictly sequential.
For strong sequential order you have to synchronize via hardware or software.

Code comments:
Highly recommend to refactor your code into methods/functions. The repetitive code is not maintainable....

Just some high level comments first:

line 543:

```pylon
threading.Timer(10.0, self.init_cam()).start()
```

You have to give a function reference to Timer. you execute self.init_cam here.

All the `RetrieveResult` calls are blocking. Like written above, this is not guaranteeing you real sequential capture times of the cameras.

Typical pattern in multicamera setups is parallel acquisition.
We have samples in the companion repo pypylon-samples: https://github.com/basler/pypylon-samples/blob/main/notebooks/multicamera_handling.ipynb

pratiksparwatwar, 2023-05-03:
Thank you for you reply.
As you have instructed I changed my code and tested for 2 days. It is running smoothly. Thank you.
Here is update code

```
from pypylon import pylon as py
import numpy as np
import cv2
import shutil
import time

NUM_CAMERAS = 6

csm1frontcam = "40271839"
csm1s1cam = "40271837"
csm1s2cam = "40271844"

csm2frontcam = "40267929"
csm2s1cam = "40271836"
csm2s2cam = "40271846"

savePath1 = "/home/rfserver/insightzz/CSM1/code/FRAME_CAPTURE/"
savePath2 = "/home/rfserver/insightzz/CSM2/code/FRAME_CAPTURE/"

def setup_cameras():
    # get the transport layer factory
    tlf = py.TlFactory.GetInstance()
    
    # create an empty list to store cameras
    cameras = []
    
    # create a device filter for Basler GigE cameras
    di = py.DeviceInfo()
    di.SetDeviceClass("BaslerGigE")
    
    # get all available Basler GigE cameras
    devices = tlf.EnumerateDevices([di,])
    
    # create an instant camera array with the desired number of cameras
    cam_array = py.InstantCameraArray(NUM_CAMERAS)
    
    # attach each camera to the instant camera array
    for i in range(NUM_CAMERAS):
        camera = py.InstantCamera(tlf.CreateDevice(devices[i]))
        camera.Open()
        cameras.append(camera)
    
    return cameras

def setup_camera_params(camera):
    # set camera parameters

    camera.GevSCPSPacketSize.SetValue(1500)          
    camera.GevSCPD.SetValue(1000)          
    camera.GevSCFTD.SetValue(1000)          
    camera.StartGrabbing(1)
    camera.AcquisitionFrameRateEnable=True
    camera.AcquisitionFrameRateAbs=3.00
    camera.ExposureTimeAbs.SetValue(15000)                
    camera.ExposureTimeRaw.SetValue(15000)             
    
    # # configure image acquisition
    # camera.StartGrabbingMax(1)

def grab_frames(cameras):
    t1 = int(time.time()*1000)
    # grab frames from each camera
    for i, camera in enumerate(cameras):
        try:
            grab_result = camera.RetrieveResult(1000, py.TimeoutHandling_ThrowException)
        except py.TimeoutException as e:
            print(f"TimeoutException: {e}")
        else:
            try:
                # convert the grab result to an OpenCV image
                img = grab_result.Array

                #converting color scheme according to camera model
                camera_name = camera.GetDeviceInfo().GetModelName()
                if camera_name == "acA1440-73gc":                
                    img = cv2.cvtColor(img, cv2.COLOR_BAYER_BG2BGR)
                elif camera_name == "acA1300-75gc":
                    img = cv2.cvtColor(img, cv2.COLOR_BAYER_BG2RGB)

                #resizing image to required resolution
                img = cv2.resize(img, (1280, 1024))

                # save the image with the camera index as the file name
                camera_id = camera.GetDeviceInfo().GetSerialNumber()  
                if camera_id == csm1frontcam:
                    cv2.imwrite(f"{savePath1}CAM1/IMG_1.jpg",img)
                    shutil.move(f"{savePath1}CAM1/IMG_1.jpg", f"{savePath1}CAM1/TMP/IMG_1.jpg")                
                elif camera_id == csm1s1cam:
                    cv2.imwrite(f"{savePath1}CAM2/IMG_1.jpg",img)
                    shutil.move(f"{savePath1}CAM2/IMG_1.jpg", f"{savePath1}CAM2/TMP/IMG_1.jpg")                
                elif camera_id == csm1s2cam:
                    cv2.imwrite(f"{savePath1}CAM3/IMG_1.jpg",img)
                    shutil.move(f"{savePath1}CAM3/IMG_1.jpg", f"{savePath1}CAM3/TMP/IMG_1.jpg")                

                elif camera_id == csm2frontcam:
                    cv2.imwrite(f"{savePath2}CAM1/IMG_1.jpg",img)
                    shutil.move(f"{savePath2}CAM1/IMG_1.jpg", f"{savePath2}CAM1/TMP/IMG_1.jpg")                
                elif camera_id == csm2s1cam:
                    cv2.imwrite(f"{savePath2}CAM2/IMG_1.jpg",img)
                    shutil.move(f"{savePath2}CAM2/IMG_1.jpg", f"{savePath2}CAM2/TMP/IMG_1.jpg")                
                elif camera_id == csm2s2cam:
                    cv2.imwrite(f"{savePath2}CAM3/IMG_1.jpg",img)
                    shutil.move(f"{savePath2}CAM3/IMG_1.jpg", f"{savePath2}CAM3/TMP/IMG_1.jpg")                

                # file_name = f"{savePath}{camera_name}_{i+1}_frame.jpg"
                # cv2.imwrite(file_name, img)
                
                # release the grab result and the camera buffer
                grab_result.Release()
                camera.StopGrabbing()
                camera.StartGrabbingMax(1)
            except Exception as e:
                print(e)

    print(f"time for one loop : {int(time.time()*1000) - t1}")

def main():
    # setup cameras
    cameras = setup_cameras()
    
    # setup camera parameters
    for camera in cameras:
        setup_camera_params(camera)
    
    # grab frames continuously
    while True:
        grab_frames(cameras)

if __name__ == "__main__":
    main()

```

I am closing this issue.

closed 2023-05-03 by pratiksparwatwar