basler/pypylon #606: weird image from pylon

pourfard, 2023-04-30:
Hi,

We have a USB camera model acA2440-35uc, but when grab image from camera there is a weird output like zebra
[image]

Sometimes there is no problem and the output image is ok.

Any idea or solution?

thiesmoeller, 2023-04-30:
This is one of the  camera testimages.
Maybe you have this configured in the user set stored in flash.

https://docs.baslerweb.com/test-images.html

pourfard, 2023-04-30:
> This is one of the camera testimages. Maybe you have this configured in the user set stored in flash.
> 
> https://docs.baslerweb.com/test-images.html

Thanks for your quick response. But test image selector is disabled.
[image]

SMA2016a, 2023-04-30:
the camera get to hot. Cool down the camera. Please check in the camera node device control for the current temperature of FPGA or Sensor

pourfard, 2023-05-01:
Thanks @SMA2016a, It seems that temperature is high in that camera.

closed 2023-05-01 by pourfard