basler/pypylon #607: GrabOne not able to do an exposure time of 2 seconds or greater

kkiekens, 2023-05-01:
If I set the exposure time to 2 seconds or greater and use:
`grab = self._camera.GrabOne(2000, pylon.TimeoutHandling_Return)`

when I check `grab.GrabSucceeded()` I get the following error:
RuntimeException: No grab result data is referenced. Cannot access NULL pointer. : RuntimeException thrown (file 'grabresultptr.cpp', line 87)

I can use up to 1.9sec without any problems. In Pylon viewer I can go up to 10sec.

SMA2016a, 2023-05-04:
better that you start the image aqquestion in continues mode and use software trigger if you need to take an image

kkiekens, 2023-05-10:
@SMA2016a Why is that a better method for grabbing a single image? Is there documentation that explains the difference between the various grab methods?

thiesmoeller, 2023-05-10:
Please check our companion repository mentioned in the readme.

Here we have a jupyter notebook demonstrating grab strategies:

https://github.com/basler/pypylon-samples/blob/main/notebooks/grabstrategies.ipynb

kkiekens, 2023-05-10:
The problem was that my timeout was set to 2s, so requesting an exposure time longer than that obviously returned nothing.

closed 2023-05-10 by kkiekens