basler/pypylon #611: Open Camera suddenly not working

duchieuphan2k1, 2023-05-17:
I use the code as the sample which is provided, include:
```
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
```
The code above working normally yesterday but today when i run exactly the same python file and code. It throws an exception:
`_genicam.RuntimeException: Could not apply configuration. Pylon::GenericException caught in OnOpened method msg=Node is not writable : AccessException thrown in node 'AcquisitionMode' while calling 'AcquisitionMode.FromString()' (file 'ValueT.h', line 85) : RuntimeException thrown (file 'AcquireContinuousConfiguration.h', line 84)`

I followed the suggestions of [xiaoxstz](https://github.com/xiaoxstz) in [this issue](https://github.com/basler/pypylon/issues/112): plug out camera, wait for a while and plug in. Then the exception is not appear anymore, but plug out/in the camera whenever this exception occurring is so inconvenient.

So is there anyway to fix this issue by python code?

SMA2016a, 2023-05-19:
would tell what camera model you are using mainly GigE or USB?

do you run multi thread application?

duchieuphan2k1, 2023-05-19:
I'm using USB and running multi thread application

duchieuphan2k1, 2023-05-22:
@SMA2016a Do you have any solution?

SMA2016a, 2023-05-22:
powercycle the camera and try to delete Genicam cache directory  e.g /home{username}/genicam_xml_cache

SMA2016a, 2023-05-22:
then make sure that your grab thread closes the camera object properly in case of error.