basler/pypylon #612: How to open camera using global shutter?

TuanHAnhVN, 2023-05-17:
Hi everyone,

I'm a newbie.
I want to capture image from basler camera (global shutter). But I cann't do it. I'm using the following code: https://github.com/basler/pypylon/blob/master/samples/grab.py

Can you give me some sample for using global shutter camera?

Thank you so much!

thiesmoeller, 2023-05-18:
What is not working?

Please post error messages or output.

If you are on windows, you have to install the pylon SDK from https://www.baslerweb.com/en/products/basler-pylon-camera-software-suite/pylon-sdks/ first, to get required drivers

TuanHAnhVN, 2023-05-18:
I am working on Linux. I can open camere through pylon viewer but python code is not :(

`An exception occurred.
Traceback (most recent call last):
  File "grab.py", line 27, in <module>
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
  File "/home/vitz/.virtualenvs/vitz_inspection/lib/python3.8/site-packages/pypylon/pylon.py", line 1592, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "grab.py", line 68, in <module>
    print(e.GetDescription())
AttributeError: 'RuntimeException' object has no attribute 'GetDescription'`

thiesmoeller, 2023-05-26:
The pylon viewer works on the the same system?