basler/pypylon #620: how to convert a grab result to a pyqt Qpixmap and adjust the frame size?

meetAndEgg, 2023-06-15:
Im trying to show frames from grap in pyqt gui, here is my current code
```python
res = self.cap.RetrieveResult(1000)
if res.GrabSucceeded():
    image = converter.Convert(res)
    frame = image.GetArray()
    frame = cv2.cvtColor(cv2.resize(frame, (self.ui.label.width(), self.ui.label.height())), cv2.COLOR_BGR2RGB)
    qframe = QImage(
        frame.data,
        frame.shape[1],
        frame.shape[0],
        frame.shape[1] * 3,
        QImage.Format_RGB888,
    )
    pixmap = QPixmap.fromImage(qframe)
    self.ui.label.setPixmap(pixmap)
    self.ui.label.repaint()
```
But I strongly doubt that this is not the best way causeI have no knowledge of video formats.
Is there a better way to convert?
Besides, I've tried resize the frame to the size of the label  (where the frame will be shown) by change the size of camera `cap.Height,cap.Width = height,width`. But an exception was thrown that the property cannot be changed. So I have to resize it using `cv2.resize(frame, (self.ui.label.width(), self.ui.label.height()))`after i get the original frame.
Can I have a better way to resize the frame?

meetAndEgg, 2023-06-16:
I've found the reason why i cannot adjust the camera size: I modify it before the camera is opened. But i still need help about what is the fastest method to convert a grab result to a pyqt Qpixmap

meetAndEgg, 2023-06-16:
Besides, I want to know that what is the best way to adjust frame size, now Im adjusting it using `cv2.resize`, but when the magnification is large(500% or more), `cv2.resize` takes a long time to complete and causes stagnation. But i see the pylon app can adjust the size quickly without stagnation

SMA2016a, 2023-07-06:
I think pylon viewer uses hardware accelerator.  You may try to use GPU support of OpenCV.