basler/pypylon #621: No device is available

pourfard, 2023-06-15:
Hi, I have a problem with new  5mp USB device. 

1- Sometimes I can grab image with abnormal FPS (5-10 FPS).
2- After a few seconds throws an exception and the camera disappears in lsusb output.
**3- Most of the time lsusb command shows the camera but it cannot find device.**

[image]

SMA2016a, 2023-06-15:
will you try your script as sudo please

pourfard, 2023-06-15:
I'm so sorry, I tested a lot of ways, but finally I tested with another cable (forth cable) and there is no problem. The problem were the broken cables.

closed 2023-06-15 by pourfard

devimauz, 2023-10-14:
sudo helped me. Thanks :)