basler/pypylon #624: Blaze 102 camera access and update IP using python

dt1729, 2023-06-23:
Hi I'm trying to create a python object by using the blaze class however I can't figure out small detail about Swig, I say that because inbuilt C++ functions are not getting called and I'm seeing errors like: 

`*** AttributeError: 'SwigPyObject' object has no attribute 'IsOpen'` 

And the C++ documentation clearly suggests that there is a function named IsOpen(), is there no SWIG binding for the same? Is blaze config not accessible via Python? 

The code that creates the IPylonDevice runs fine. Here is the code for the same:

`

from pypylon import pylon 
dev1 = pylon.TlFactory.GetInstance().CreateFirstDevice(pylon.CDeviceInfo().SetDeviceClass('BaslerGTC/Basler/GenTL_Producer_for_Basler_blaze_101_cameras'))

`

Could you please tell me if Python is good for this task or shall I try the C++ api for configuring IP addresses?

Thanks!

SMA2016a, 2023-06-25:
is the camera available  in  pylonviewer and configuration tool?

dt1729, 2023-06-27:
not in pylon viewer but IP configuration tool works fine.

SMA2016a, 2023-06-27:
do you have DHCP server on your system?  what is the IP of the camera and NIC?

SMA2016a, 2023-06-27:
Hi Simply install pylon  Supplementary Package  for blaze and then you can simply open the camera as usual 

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

!! you need to call get bufer instead of get array if you have the grabresult from blaze camera.  e.g

grabResult.Buffer

dt1729, 2023-06-27:
Can you link this package I'll try it first thing today thanks!