basler/pypylon #625: Can't detect multiple cameras

AlexRoss-WHS, 2023-06-26:
Hi. I'm trying to set up a System with multiple acA2440-gc GigE-cameras. For a start I am only trying to detect the cameras with the following code:
```python
from pypylon import pylon

tlf = pylon.TlFactory.GetInstance()
devs = tlf.EnumerateDevices()

for d in devs:
    print(d.GetModelName(), d.GetSerialNumber())
```
When I try it with only one camera connected it works (Tested with different cameras and ports). But when I try to detect multiple cameras at once only one is detected. 

I had to use the pylon GigE-configurator with `PylonGigEConfigurator auto-all` before being able to detect even one camera and have to do so after every reboot (sometimes multiple times before it works). The GigE-configurator seems to detect all cameras.

I am using Ubuntu 22.04.2 LTS. 

Is there anything I am missing here? How can I best troubleshoot this?

AlexRoss-WHS, 2023-06-27:
Hi @thiesmoeller ,

thanks for the response. I am connecting the cameras directly to the ports of the PC. I have not assigned any adresses to any ports from ubuntu. Should I have done so?
If the tool only temporarily changes the configurations, how do I change them permanently?

I actually found a strange workaround. If I do `sudo nmcli dev set <device_name> managed no` followed by `sudo nmcli dev set <device_name> managed yes` for all devices multiple times after every reboot, all cameras are eventually detected. I don't really know much about networks, so I have no idea as to why this works.

SMA2016a, 2023-07-01:
yes, you need to set IP addresses properly to work with multiple cameras. Also you need to set the IP of the network card and bring them in separate network using subnet mask too.

e.g  
NIC 1   192.168.1.1/255.255.255.0    camera   192.168.1.2/255.255.255.0
NIC 2    192.168.2.1/255.255.255.0    camera   192.168.2.2/255.255.255.0
etc...

closed 2023-07-03 by AlexRoss-WHS