basler/pypylon #626: low fps and crahes

ligthmare, 2023-06-26:


ligthmare, 2023-06-26:
Hi everyone,

I’m trying to capture a video stream from a basler gige 5mp camera.

I’m using a jetson nano 4gb card I’ve installed Pypylon and opencv

I can open the camera and display it but the fps are very low with crashes.

If the display window is small it seems ok (can’t do much better) but when I enlarge the window it slows down a lot and crashes.

I tried to reduce the bandwidth (the camera consumes 102Mb/s) but same result.

on the other hand, when I reduce the number of pixels to be displayed to 500*200px, for example, I get great fluidity back.

I’m completely dry, I don’t know if it’s the nano’s network card that’s saturated (I don’t believe it) or the CPU or RAM capacity. But when I look at the RAM and CPU usage in the task manager, I’m only at about 50% usage.

I used the sample code find here :

```
 from pypylon import pylon
 import cv2
 
 # conecting to the first available camera
 camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
 
 # Grabing Continusely (video) with minimal delay
 camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
 converter = pylon.ImageFormatConverter()
 
 # converting to opencv bgr format
 converter.OutputPixelFormat = pylon.PixelType_BGR8packed
 converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
 
 while camera.IsGrabbing():
     grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
 
     if grabResult.GrabSucceeded():
         # Access the image data
         image = converter.Convert(grabResult)
         img = image.GetArray()
         cv2.namedWindow('title', cv2.WINDOW_NORMAL)
         cv2.imshow('title', img)
         k = cv2.waitKey(1)
         if k == 27:
             break
     grabResult.Release()
     
 # Releasing the resource    
 camera.StopGrabbing()
 
 cv2.destroyAllWindows()
```

thiesmoeller, 2023-06-27:
One hint regarding processor load in your setup, would be to configure the PixelFormat of your camera.
Per factory default ( in case you have a color camera ) the camera outputs 8-bit Bayer raw. To convert this to BGR is a CPU intensive operation in your code.
This get's more effective, if you configure the camera to output a YUV format.

ligthmare, 2023-06-27:
Hi thiesmoeller , I already change the Pixelformat by changing the code directly or configure camera by pylonviewer. Same result 
So could(need) i use the GPU to do this kind of calcul? possible to do it in python? i read some things about cuda but i found only in C++

thiesmoeller, 2023-06-27:
To lower the load on your system you should also follow the overall network setup documentation for your camera 

https://docs.baslerweb.com/network-configuration-(gige-cameras)#changing-the-network-adapter-properties-linux

ligthmare, 2023-06-28:
seem not good :(

ligthmare, 2023-06-28:
I really don't understand what is the problem 

My code: 
`from pypylon import pylon
import cv2
 
 # conecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
 
 # Grabing Continusely (video) with minimal delay

camera.GevSCPSPacketSize.SetValue(9000)

camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
converter = pylon.ImageFormatConverter()
 
 # converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
 
while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
 
    if grabResult.GrabSucceeded():
         # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.namedWindow('title', cv2.WINDOW_NORMAL)
        cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()
     
 # Releasing the resource    
camera.StopGrabbing()
 
cv2.destroyAllWindows()`

Error : 
`simec@simec-desktop:~/Desktop$ python3 test.py 
Traceback (most recent call last):
  File "test.py", line 9, in <module>
    camera.GevSCPSPacketSize.SetValue(9000)
  File "/home/simec/.local/lib/python3.6/site-packages/pypylon/genicam.py", line 1898, in SetValue
    return _genicam.IInteger_SetValue(self, Value, Verify)
_genicam.AccessException: Node is not writable. : AccessException thrown in node 'GevSCPSPacketSize' while calling 'GevSCPSPacketSize.SetValue()' (file 'IntegerT.h', line 77)
terminate called without an active exception
Aborted (core dumped)
`

I can increase the packet size but you why this could not be the CPU use? 

I try to install also Gstreamer plugin without result also  :
`gst-launch-1.0 pylonsrc ! videoconvert ! autovideosink`

error : 
`Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstPylonSrc:pylonsrc0: Failed to create buffer.
Additional debug info:
../ext/pylon/gstpylonsrc.cpp(993): gst_pylon_src_create (): /GstPipeline:pipeline0/GstPylonSrc:pylonsrc0:
The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e., the network adapter, switch, or Ethernet cable. Buffer underruns can also cause image loss. To fix this, use the pylonGigEConfigurator tool to optimize your setup and use more buffers for grabbing in your application to prevent buffer underruns.
Execution ended after 0:00:00.892215426
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...`

thiesmoeller, 2023-06-28:
You are missing the `camera.Open()` before any feature access to the camera. See my example.