basler/pypylon #627: Capturing Images at Specific Points during Exposure Timeframe

eksiemin, 2023-06-27:
Hello Basler team,

I am currently working with the Basler a2A2600-64umBAS camera and I am interested in capturing images at specific points during the complete exposure timeframe. I would like to know if there are any features or functionalities available in Basler cameras or the Pylon SDK that would allow me to achieve this.

Specifically, I would like to capture images at precise points or phases within the exposure timeframe. For example, capturing an image at the beginning, middle, and end of the complete exposure duration. I am aware that interrupting or modifying the exposure timeframe itself is not feasible, but I'm wondering if there are any techniques or features that would enable me to achieve this level of control over image capture timing.

If there are any APIs, or SDK functionalities that support capturing images at specific phases during the exposure timeframe, I would appreciate any guidance, documentation, or examples you can provide.

Thank you in advance for your assistance.

Best regards

thiesmoeller, 2023-06-27:
Hi @eksiemin,

I understand, what you describe as the request to readout the image multiple times during one exposure timeframe.
If this would be available, it is a capability of the image sensor itself.

The GMAX2505 in your camera model is not capable of doing this. Most image sensors are not capable of doing this, as reading out the charge of a pixel means resetting it. 
Image sensors that don't have the reset-on-readout behavior can do multiple NDRO ( Non destructive readout ). You'll find it typically on higher end scientific image sensors.

To help you find a solution with _your_ camera, can you describe the problem that you want to solve?

eksiemin, 2023-06-27:
Hi @thiesmoeller,

Thank you very much for the feedback!

Actually what I really want is to get at least 2-3 images during one single exposure time. Because of the conditions of my process in each image (depending on the xp. time) I see different aspects of the object and since I work on a moving line (my object is moving) it is only valuable for if I could get my shorter and shortest exp. time frames within the same exp. timeframe. 

Just an example: Let us imagine that I have an exposure time of 5000 microseconds during a shut. I also would like to get the images when it is by the 3000th microseconds or 4000th microseconds. So, 3 images from 1 shut.

Is it somehow possible?

Thank you / Best Regards
Emin

thiesmoeller, 2023-06-28:
Hm. 

So you want three images.
All starting their exposure at the exact same time but with different exposure times.

Image 1 3ms
Image 2 4ms
Image 3 5ms

So image number three will be the brightest and image 1 the darkest image ?
And image 1 has the least and image 3 the strongest motion blur due to your conveyor belt?

eksiemin, 2023-06-28:
Hello Thies,

That is exactly what we need and any solution proposal will be appreciated a lot!
Thank you in advance.

BR
Emin

thiesmoeller, 2023-06-30:
The camera as you have it now, is not capable to do non-destructive-readout.

One option on above conveyor belt, and if your field of view is a lot larger than your object is image stacking.

You acquire a number of images with shorter exposure times. To reach a higher framerate, you have to reduce the ROI and use the burst acquisition feature of your camera.

This image stack can now shifted ( to compensate the motion ) and added on top of each other.
With this setup you can get your different exposure times, just stretched over space and time.

[image]

We can also talk about customized options to solve your requirements ( contact Basler support / depending on project volume ).