basler/pypylon #628: High temperature

pourfard, 2023-06-28:
Hi,

I have ace2 5 mega pixels USB3 camera and I want to use it in outdoor. I have used a CCTV housing with sun shield and currently weather temperature is about 43 degrees in our city. The cameras temperature is about 80 degrees and it cannot capture image. I have used two fans and heat sinks to reduce the temperature and I just grab only 10 frames per second. Is there any solution or idea to grab frames in this condition?

Thanks

SMA2016a, 2023-06-28:
I think only active cooling will help you since the environment temperature is quite high.

pourfard, 2023-07-01:
We had some LEDs in front of the camera's housing and we turned all them off and now the camera's temperature is about 68 degrees.

closed 2023-07-01 by pourfard