basler/pypylon #630: Finding cameras by IP address

eduardorojasramirez, 2023-07-04:
Hello,

Read the documentation and managed to create some code to find all my cameras and grab frames from them, to then export them to PNG format. Is there a way to find a camera using IP address as a parameter or maybe filtering it by a property related to it.

Here is the code i currently have by the way, I know it is a little rough, this is still a PoC.
`from pypylon import pylon
import uuid

img = pylon.PylonImage()
tlf = pylon.TlFactory.GetInstance()
devs = tlf.EnumerateDevices()
cams = pylon.InstantCameraArray(len(devs))
index = 0
for d in devs:
    print(d.GetModelName(), d.GetSerialNumber())
    cams[index].Attach(tlf.CreateDevice(d))
    cams[index].Open()
    cams[index].StartGrabbing()
    with cams[index].RetrieveResult(1000) as result:
        # Calling AttachGrabResultBuffer creates another reference to the
        # grab result buffer. This prevents the buffer's reuse for grabbing.
        img.AttachGrabResultBuffer(result)
        filename = str(uuid.uuid4())+ ".png"
        img.Save(pylon.ImageFileFormat_Png, filename)
        # In order to make it possible to reuse the grab result for grabbing
        # again, we have to release the image (effectively emptying the
        # image object).
        img.Release()
    cams[index].StopGrabbing()
    cams[index].Close()
    index +=1`

SMA2016a, 2023-07-06:
I am sure if I have understood you correctly.

you can create a device object and set the known property and pass it to 

ip_address = "192.168.3.10"
cam_info = pylon.DeviceInfo()
cam_info.SetPropertyValue('IpAddress', ip_address)
cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice(cam_info))

you just need to use the pre-set device info object instead of d

cams[index].Attach(tlf.CreateDevice(d))

eduardorojasramirez, 2023-07-06:
Hello, thanks but not exactly what I was looking for. I am trying to find a camera by using the IP address as filter. I just managed to find a function using the documentation found in Pylon Viewer on the features settings, now my code looks like this. 

```
from pypylon import pylon
import socket, struct, os, uuid

def createPictures(cameraList):
    img = pylon.PylonImage()
    tlf = pylon.TlFactory.GetInstance()
    devs = tlf.EnumerateDevices()
    cams = pylon.InstantCameraArray(len(devs))
    index = 0
    paths = []
    for d in devs:
        # print(d.GetModelName(), d.GetSerialNumber())
        cams[index].Attach(tlf.CreateDevice(d))
        cams[index].Open()
        ip_address = socket.inet_ntoa(struct.pack('!L',cams[index].GevPersistentIPAddress.GetValue()))
        # print(ip_address)
        if ip_address in cameraList:
            cams[index].GainAuto.SetValue('Continuous')
            cams[index].StartGrabbing()
            with cams[index].RetrieveResult(1000) as result:
                # Calling AttachGrabResultBuffer creates another reference to the
                # grab result buffer. This prevents the buffer's reuse for grabbing.
                img.AttachGrabResultBuffer(result)
                filename = "C:\\temp\\" + str(uuid.uuid4())+ ".png"
                img.Save(pylon.ImageFileFormat_Png, filename)
                path = os.path.abspath(filename)
                # print(path)
                paths.append(path)
                # In order to make it possible to reuse the grab result for grabbing
                # again, we have to release the image (effectively emptying the
                # image object).
                img.Release()
            cams[index].StopGrabbing()
        cams[index].Close()
        index +=1
    return paths
```

Still a little rough, but does what I need. The cameraList object is an array of IP addresses. Will go ahead and close the issue, since the question has found an answer, thanks again for the reply.

closed 2023-07-06 by eduardorojasramirez