basler/pypylon #631: Stop RetrieveResult

AlexRoss-WHS, 2023-07-05:
Hi, I have a quick question. Is it possible to break out of `camera.RetrieveResult(pylon.waitForever)`, so that I can stop the grabloop without waiting for a frame?

SMA2016a, 2023-07-06:
why do not use the onImagegrab call back?  

# Example of an image event handler.
class SampleImageEventHandler(pylon.ImageEventHandler):
    def OnImageGrabbed(self, camera, grabResult):
        print("CSampleImageEventHandler.OnImageGrabbed called.")
        cameraContextValue = grabResult.GetCameraContext()
        print("Camera ID ", cameraContextValue)
        print("image ID ", grabResult.BlockID)
        print("image Time Stamp ", grabResult.TimeStamp)
        print()
        print()

        # For demonstration purposes only, register another image event handler.
        cam.RegisterImageEventHandler(SampleImageEventHandler(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)

AlexRoss-WHS, 2023-07-10:
I tried but the `OnImageGrabbed` hook only seems to be called if i also still call `cam.RetrieveResult()` in a loop. Am I missing something here?

thiesmoeller, 2023-07-10:
;-) this is a side effect only.

You have to tell StartGrabbing that you want to use the background loop

See details here https://github.com/basler/pypylon-samples/blob/main/notebooks/grabstrategies.ipynb

AlexRoss-WHS, 2023-07-10:
What if I don't want to use the background loop? Is there any way doing this within the current thread?

thiesmoeller, 2023-07-10:
If you don't want to work with threading .. 

Use normal foreground  RetrieveResult 
set a small timeout and have a loop around it, that ignores the incoming timeout exceptions until your global abort criteria is met. This will just create some polling overhead to your program

AlexRoss-WHS, 2023-07-10:
I am using many cameras at the same time. Can I just start them all with their own background loop? Will the `OnImageGrabbed` hook be executed in the background loop or the main thread? My current setup uses multiprocessing similar to your answer in [this issue](https://github.com/basler/pypylon/issues/513#issuecomment-1346405311) and works fine, except that I can only stop after an image is taken.