basler/pypylon #632: How to Record a Video and Preview the Screen in Real Time

Achuan-2, 2023-07-06:
I am using BASLER acA1440 - 220 um.
I want to use the camera to record a Video and preview the screen in real time just like `pylon viewer`
Could you give me some help or advice?

[image]

Although I used C++programming and the preview screen got stuck, I couldn't stop recording halfway like the official software did
[image]

SMA2016a, 2023-07-06:
The pylon viewer does not show the video, instead of it shows the grabbed image, same like the sample code demonstrates. But the pylon viewer a multi threading application.

thiesmoeller, 2023-07-11:
For a camera that can generate 220fps and a display that'll show at 60fps youl always need a decoupler that will subsample the incoming stream.