basler/pypylon #635: Light triggering using raspberry pi

shrinand1996, 2023-07-18:
Hello, I am trying to strobe a light whenever I get a signal from my proximity sensor. The code is as follows:
if (trigger == 1):
     turn the lights on (GPIO HIGH)
     grabresult.RetrieveResult(...)
     if (grabresultretrieveResult Succeed):
         convert 
         turn the lights off (GPIO LOW)

The code is strobing the light and capturing the image. 
The issue is that the images are severly/ partially exposed. To overcome this issue, I tried to add a delay of 2 s (which is a lot but it will still remain underexposed. )
Now i dont understand that if i am giving the command to turn on the lights before capture and turning it off after capture, why are the images underexposed?
And how can i get rid of the delay?

Since i am using a pi, I dont need an external light controller to strobe the light. 

How to fix this?

SMA2016a, 2023-07-20:
why do not stobe the light via cameras GPIO?  and exposure active signal?

shrinand1996, 2023-07-24:
I am not using the camera GPIO as i am using a dart camera and dont have the I/O board. Since i have full control of the light and trigger in raspberry pi, i thought ill take images directly trough pypylon and control the lights as well.

HighImp, 2023-10-17:
If you are sure that the light is not controlled by pwm,
ensure that the exposure starts after the light is on. Try to use SoftwareTrigger after the Lights are on or simply use: 
grab_result = camera.GrabOne(<timeout>) 
instead of start/stop grabbing