basler/pypylon #638: Pylon Apps do not start up | Arm64 Linux

matt-VR, 2023-07-26:
Hi there, i have successfully installed pylon7.3 deb package on my Jetson Nano ARM64. However, if i click on the IP Configurator icon or the pylon Viewer icon, nothing happens. The apps do not start up. Any ideas how to fix that?

thiesmoeller, 2023-07-26:
Start from a terminal and post the output
Binaries are in /opt/pylon/bin

matt-VR, 2023-07-26:
Thanks for your reply. Here's the output:

ipconfigurator: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /opt/pylon/bin/../lib/pylonviewer/libPylonViewerComponents.so.7)
ipconfigurator: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /opt/pylon/bin/../lib/Qt/lib/libQt5Widgets.so.5)
ipconfigurator: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /opt/pylon/bin/../lib/Qt/lib/libQt5Gui.so.5)
ipconfigurator: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /opt/pylon/bin/../lib/Qt/lib/libQt5Core.so.5)
ipconfigurator: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /opt/pylon/bin/../lib/Qt/lib/libQt5Core.so.5)

PS. i have glibc 2.27 on my system

thiesmoeller, 2023-07-27:
You have to run an older pylon version. Pylon is linked to system libraries that are newer than the ones on your Jetson Nano.

The Version on your system is Ubuntu 18.04 (https://developer.nvidia.com/jetpack-sdk-464), which is using very old system libraries.

One option to run with newest pylon, is to run in a docker container on your nano.

SMA2016a, 2023-07-28:
or update the image to Ubuntu 20.

https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image

matt-VR, 2023-07-28:
Thank you very much for your help. Pylon 6.2 works on my 18.04 and should be ok for now. If I ever upgrade to 20.04 or try it in docker I'll post it here

hvn2, 2024-02-29:
I have installed pylon7.4.0 arm64 on Raspberry Pi 5, Debian OS 64 bits (bookworm) and got the same problem (nothing happen when click and run pylonviewer). Running pylonviewer on terminal (./pylonviewer) results: segmentation fault. How to fix this problem? 
Thanks.

SMA2016a, 2024-02-29:
try to reinstall the pylon using apt

apt install ./*.deb

hvn2, 2024-03-01:
> try to reinstall the pylon using apt
> 
> apt install ./*.deb
> 
> copy the pylon debian package in working directory
> 
> if you build the sample code, run the application? for ex. grab.cpp
Thanks for you reply. The problem has not been solved.
I downloaded installation package (to Downloads folder), extract it into a folder (pylon_setup) inside Downloads
I changed the working directory to Downloads/pylon_set up and run sudo apt install ./pylon*.deb ./codemeter*.deb
Running a python sample code got error no camera detect.
Note that: the camera works fine  on raspberry pi 4, debian 32 bits (bulleye) and pylon 6.1.