basler/pypylon #641: Windows fatal exception: access violation

ridvanozdemir, 2023-08-08:
Hello, I have a Python Qt interface and 2 basler cams system. one of them is working on free run mode and the other one is working with hardware trigger. But like after working 1 hour I get an error like this:

Can you help me with that?

Windows fatal exception: access violation

Thread 0x00000d34 (most recent call first):
  File "C:\Users\Kadir\AppData\Roaming\Python\Python310\site-packages\pypylon\pylon.py", line 3495 in RetrieveResult
  File "c:\ecg\ecg01_3d_2d_v28.py", line 1277 in run

Thread 0x00002264 (most recent call first):
  File "C:\Users\Kadir\AppData\Roaming\Python\Python310\site-packages\pypylon\pylon.py", line 3495 in RetrieveResult
  File "c:\ecg\ecg01_3d_2d_v28.py", line 1277 in run

Main thread:
Current thread 0x000020f8 (most recent call first):
  File "c:\ecg\ecg01_3d_2d_v28.py", line 959 in ImageUpdateSlot
  File "c:\ecg\ecg01_3d_2d_v28.py", line 1412 in <module>
  File "C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\py3compat.py", line 356 in compat_exec
  File "C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 469 in exec_code
  File "C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 611 in _exec_file
  File "C:\ProgramData\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 524 in runfile
  File "C:\Users\Kadir\AppData\Local\Temp\ipykernel_10752\365309949.py", line 1 in <module>

Restarting kernel...

SMA2016a, 2023-08-13:
what camera interface  in the game? if you run your hardware trigger with Pylon viewer, do you see the same error after 1h?

ridvanozdemir, 2023-08-13:
I didn't try it on Pylon, I use PyQt5 as a GUI. and this is code.

`class Worker1(QtCore.QThread):
    ImageUpdate = QtCore.pyqtSignal(QImage, QImage, QImage, int)

    #fault_signal = QtCore.pyqtSignal(int)
    # def __init__(self, camera, name):
    #     threading.Thread.__init__(self)
    #     self.camera = camera
    #     self.name = name
    def __init__(self, camera):
        super(Worker1, self).__init__()
        self.camera = camera

        
    def run(self):
        #self.UI = UI() 
        self.ThreadActive = True

        # print("Starting Thread...")
        # grab_times = []

        # used to record the time when we processed last frame
        prev_frame_time = 0          
        # used to record the time at which we processed current frame
        new_frame_time = 0
        
        global total_product, Pic, Pic2, Pic3, Qt_NumNokProduct, NumProduct, Qt_NumProduct 
        global reject_2D_, image3D, reject_3D_, reject_Img_, error_3D, error_2D, fault_class, fault_class3D, fault_class2D, image_
        Pic2_OK=False
        self.camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
        print("start grabbing")
        while self.camera.IsGrabbing():
            try:
   
                with self.camera.RetrieveResult(9995000, pylon.TimeoutHandling_ThrowException) as res:
                    
                    if res.GrabSucceeded():
                        #print("grab succeed")
                        #grab_times.append(res.TimeStamp)
                        #image1=res1.Array()                        
                        image = converter.Convert(res)
                        image = image.GetArray()

                        if self.camera.GetDeviceInfo().GetSerialNumber()=="24604775":
                            #print("77li cam ok")
                            
                            #outputs image, reject_3D, reject_Img, fault_class3D
                            image3D, reject_3D_, reject_Img_, fault_class3D=check3D(image)                                                                                          
                                                        
                        elif self.camera.GetDeviceInfo().GetSerialNumber()=="24609272":
                            #print("66lu cam ok")
                            Qt_NumProduct+=1   
                            #2D control
                            # image, img_cropped, reject_2D, fault_class2D
                            image_, image2D, reject_2D_, fault_class2D=check2D(image)
                            
                            # #update reject 2d image
                            img_cropped = cv2.cvtColor(image2D, cv2.COLOR_BGR2RGB)
                            Pic2= QImage(img_cropped.data, img_cropped.shape[1], img_cropped.shape[0], img_cropped.strides[0], QImage.Format_RGB888)

                            #update product 2d image
                            img_2d = cv2.cvtColor(image_, cv2.COLOR_BGR2RGB)
                            Pic3= QImage(img_2d.data, img_2d.shape[1], img_2d.shape[0], img_2d.strides[0], QImage.Format_RGB888)

                            #update 3d cam's reject image
                            frame = cv2.cvtColor(reject_Img_, cv2.COLOR_BGR2RGB)                            
                            Pic= QImage(frame.data, frame.shape[1], frame.shape[0], frame.strides[0], QImage.Format_RGB888)
                            
                            if (fault_class3D>0):
                                fault_class=fault_class3D
                            else:
                                fault_class=fault_class2D
                            
                            
                            self.fault_count(fault_class)
                            
                            try:
                                self.ImageUpdate.emit(Pic, Pic2, Pic3, fault_class)
                                
                            except Exception as em:
                                print(em)
                                print("güncelleme yapılamıyor!")
                                
                                # Check if the file exists before appending a new line
                                if os.path.isfile(txtPathName):
                                    # Open the file in "append" mode
                                    with open(txtPathName, 'a') as txtfile:
                                        txtfile.write(datetime.now().strftime("%Y_%m_%d-%H_%M_%S") + " : güncelleme yapılamıyor! " + '\n')
                                    txtfile.close()    
                                else:
                                    # Print an error message to the console if the file doesn't exist
                                    print('File does not exist.')
                                self.stop_worker_1()
                                self.start_worker_1()
                            

                                                       
                                                                    
            except Exception as e:
                print(e)
                self.camera.StopGrabbing()
                # Check if the file exists before appending a new line
                if os.path.isfile(txtPathName):
                    # Open the file in "append" mode
                    with open(txtPathName, 'a') as txtfile:
                        txtfile.write(datetime.now().strftime("%Y_%m_%d-%H_%M_%S") + " : kamera grabbing durdu " + '\n')
                    txtfile.close()    
                else:
                    # Print an error message to the console if the file doesn't exist
                    print('File does not exist.')
                

                break
  
    def stop(self):
        self.ThreadActive = True
        self.quit()

       

app = QtWidgets.QApplication(sys.argv)
UIWindow = UI()
MainWindow = QMainWindow()

UIWindow.show()
#sys.exit(app.exec_())
app.exec_()   `