basler/pypylon #643: How to capture images from 2 different cameras with 2 different  triggers.

PaVaNTrIpAtHi, 2023-08-14:
Hello
I have a requirement where I have 2 conveyor lines running. I have 2 triggers attached to 2 different cameras. I am able to capture particular images in pylon viewer. The hardware triggers  and camera are doing their job. Just want a reference as to how to do this in python. 

Thanks

PaVaNTrIpAtHi, 2023-08-16:
What If I want to start all my cameras with a particular configuration set?

thiesmoeller, 2023-08-18:
The camera array is just a simplification to handle more than one camera. You can do this one by one too.

If your triggers are HW signals, connected to cam 1/2 and cam 3/4 respectively the right cameras will start.
You setup exactly the same configuration as in pylonviewer.

The order of images you receive is the order of trigger signals you get.

For overall grab strategies see also the companion repository pylon samples: https://github.com/basler/pypylon-samples/blob/main/notebooks/grabstrategies.ipynb

PaVaNTrIpAtHi, 2023-08-22:
Thanks for the information.
 [https://github.com/basler/pypylon/blob/master/samples/grabmultiplecameras.py](url) this file did help.

I will post the code which contains convertor and user set selector attached with cameras(could be of help to someone someday) and close the issue.

Thanks

PaVaNTrIpAtHi, 2023-08-25:
Please note this is a sample code which works fine on my system, but I haven't provided the entire class structure. Consider the syntax of code and you will be able to get the work done.

`# Get all attached devices and exit application if no device is found.

#code to initialize cameras with specific default settings
        

```
        self.devices = self.tlFactory.EnumerateDevices()
        maxCamerasToUse = 4
        try:
            if len(self.devices) == 0:
                self.top_device_absent = True
            # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
            self.cameras = pylon.InstantCameraArray(min(len(self.devices), maxCamerasToUse))
            print(self.cameras)
            l = self.cameras.GetSize()
            print("cam thread cam size ",l)

            # Create and attach all Pylon Devices.
            for i, cam in enumerate(self.cameras):
                print(cam)
                cam.Attach(self.tlFactory.CreateDevice(self.devices[i]))
                cam.Open()

                self.PixelFormat = cam.PixelFormat.GetValue()

                # Set converter.OutputPixelFormat
                # self.converter = pylon.ImageFormatConverter()
                if(cam == 'Mono8'):
                    self.converter.OutputPixelFormat = pylon.PixelType_Mono8
                else:
                    self.converter.OutputPixelFormat = pylon.PixelType_BGR8packed
                self.converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

                cam.UserSetSelector = "UserSet2"
                cam.UserSetLoad.Execute()

                # Print the model name of the camera.
                print("Using device ", cam.GetDeviceInfo().GetModelName())

            camera_info_dict = {}

            # store a unique number for each camera to identify the incoming images
            for idx, cam in enumerate(self.cameras):
                camera_serial = cam.DeviceInfo.GetSerialNumber()
                camera_info_dict[idx] = cam.DeviceInfo.GetSerialNumber()
                print(f"set context {idx} for camera {camera_serial}")
                cam.SetCameraContext(idx)`

#code to start grabbing based on triggers

       ```
self.cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

        while self.__running.isSet():
            self.__flag.wait()
            try:
                print(" while start ---------------------------------------")
                camera_grab_st = time.time()
                
                # Grab c_countOfImagesToGrab from the cameras.
                for i in range(10000000):
                    print(i, " count")
                    if not self.cameras.IsGrabbing():
                        break

                    grabResult = self.cameras.RetrieveResult(pylon.waitForever, pylon.TimeoutHandling_Return)
                    
                    # When the cameras in the array are created the camera context value
                    # is set to the index of the camera in the array.
                    # The camera context is a user settable value.
                    # This value is attached to each grab result and can be used
                    # to determine the camera that produced the grab result.
                    cameraContextValue = grabResult.GetCameraContext()
                    # set context 0 for camera 24153744
                    # set context 1 for camera 24640129
                    # set context 2 for camera 24678966
                    # set context 3 for camera 40200296
                    
                    print("Camera ", cameraContextValue, ": ", self.cameras[cameraContextValue].GetDeviceInfo().GetSerialNumber())

                    # Now, the image data can be processed.
                    print("GrabSucceeded: ", grabResult.GrabSucceeded())
                    # print("SizeX: ", grabResult.GetWidth())
                    # print("SizeY: ", grabResult.GetHeight())
                    if grabResult.GrabSucceeded():
                        #line 1 combo is 0,3
                        if cameraContextValue==0 :
                            self.top_img_rgb = self.converter.Convert(grabResult).GetArray()
                            self.top_count +=1
                            self.t1 = 1
                            print(self.top_count, " top valid count")
                            # print(top_img_path)
                            # cv2.imwrite(top_img_path, self.top_img_rgb)
                        elif cameraContextValue==3:
                            self.side_img_rgb = self.converter.Convert(grabResult).GetArray()
                            self.side_count +=1
                            self.s1 = 1
                            # print(side_img_path)
                            print(self.side_count, " side valid count")
                            # cv2.imwrite(side_img_path, self.side_img_rgb)

                        #line 2 combo
                        elif cameraContextValue==2:  #top cam line 2
                            self.top_img_rgb_2 = self.converter.Convert(grabResult).GetArray()
                            self.top_count +=1
                            self.t2 = 1
                            
                            print(self.top_count, " top valid count")
                            # print(top_img_path)
                            # cv2.imwrite(top_img_path, self.top_img_rgb)
                        else:
                            self.side_img_rgb_2 = self.converter.Convert(grabResult).GetArray()
                            self.side_count +=1
                            self.s2 = 1
                            # print(side_img_path)
                            print(self.side_count, " side valid count")
                            # cv2.imwrite(side_img_path, self.side_img_rgb)

```

closed 2023-08-25 by PaVaNTrIpAtHi