basler/pypylon #644: camera link + frame grabber gpio OUT issue

emilyjanedennis, 2023-08-22:
hey folks,

**Is there a way to trigger a TTL programmatically in python? Reasoning below: tl;dr I want to debug with our oscilloscope and can't get any signal under any circumstances and documentation for our camera is hard to find or doesn't exist.**

I'm trying to get our microenable 5 marathon ACL framegrabber front GPIOs to output a TTL-like pulse using the rising trigger from either exposure or acquisition (doesn't matter as long as we're getting a pulse for each frame) on our Basler ace acA2040-180kmNIR which is connected via 2 camera link cables. I've hooked up an oscilloscope to our GPIO out and see nothing regardless of settings, and I've tried triggering acquisition via python and microdisplay X to no avail. I do have a working python script that saves video from the camera, so we have good connection, etc. 

The online documentation for Line Source, etc. doesn't have our camera model, and I've tried using pylon, microdisplay X parameters and can't find either sufficient documentation to be sure I'm doing things right. So, I'm hoping there's a way to use python and get the framegrabber to send a TTL via front GPIOs. 

Any/all help much appreciated. I also contacted Basler support a week ago and still haven't heard back, seems like this is more frequently updated. Thanks! -Emily

BryanLiou, 2024-03-19:
Hello Emily, 

I also encountered the same problem, have you solved it?

Thanks!
Bryan

emilyjanedennis, 2024-03-19:
Unfortunately no, we ended up buying the TYL board and it's still difficult
and basler help has been extremely unhelpful

cpt-wojtech, 2024-04-05:
Hi @emilyjanedennis would it be an option for you to use the cameras GPIO outputs? They are well described in the cameras manual, which is not on the Online Product Documentation but can be downloaded here 
https://www2.baslerweb.com/de/downloads/downloads-dokumente/basler-ace-camera-link-users-manual/
You could use the ExposureActive signal on one of the Lines to get an high signal every time the exposure of an image starts.

The second hurdle that I can see is that pypylon is not necessarily the best option to work with the framegrabber.  Maybe take a look at the Python Wrapper for the Framegrabber SDK https://docs.baslerweb.com/frame-grabbers/python-wrapper
They are different API's: the pypylon API is good for CXP cameras and CXP Framegrabbers, while the Python-Wrapper is in general better for Framegrabbers. 

I can try to check if there is a possibility to use the Outputs of the grabber using the cameras ExposureActive signal.

floesche, 2024-10-10:
I helped with getting a TTL signal out of the framegrabber and wanted to document what we did, in case someone else tries to solve a similar issue.

The framegrabber can load different ["applets"](https://docs.baslerweb.com/frame-grabbers/acquisition-applets-for-marathon-acl), which programs the FPGA inside the [Basler microEnable 5 marathon ACL](https://www.baslerweb.com/en-us/shop/microenable-5-marathon-acl/). While Basler provides a [visual programming language](https://www.baslerweb.com/en/software/visualapplets/) for the FPGA, I was able to use an existing applet to get a TTL.

For the combination of camera and framegrabber, I chose the "Acq_SingleBaseAreaGray" applet. I used the [python wrapper of the Basler Framegrabber SDK](https://docs.baslerweb.com/frame-grabbers/python-wrapper) to interact with the applet. Following the [documentation for the applet](https://docs.baslerweb.com/frame-grabbers/pdf/me5-ma-acl/AW00176901_Acq_SingleBaseAreaGray_fglib.pdf), I used the following snippet to switch the camera to software trigger (line 2), set the maximum triggers per second to 200 (line 3), initialized a pulse generator with a certain width and delay (lines 4…5), and set the output to the front GPIO (line 6). Finally I make the trigger active (line 7):

```python
import SiSoPyInterface as s
s.Fg_ParameterWithInt(frame_grabber, s.FG_AREATRIGGERMODE, s.ATM_SOFTWARE, port)
s.Fg_ParameterWithDouble(frame_grabber, s.FG_TRIGGER_FRAMESPERSECOND, 200, port)
s.Fg_ParameterWithDouble(frame_grabber, s.FG_TRIGGER_PULSEFORMGEN0_WIDTH, width, port)
s.Fg_ParameterWithDouble(frame_grabber, s.FG_TRIGGER_PULSEFORMGEN0_DELAY, delay, port)
s.Fg_ParameterWithInt(frame_grabber, s.FG_TRIGGEROUT_SELECT_FRONT_GPO_0, ⏎
    s.CAM_A_PULSEGEN0, port)
s.Fg_setParameterWithInt(frame_grabber, s.FG_TRIGGERSTATE, s.TS_ACTIVE, port)
```

After this setup, a TTL with a certain `width` and delayed by `delay` can be triggered by the following lines:

```python
err = s.Fg_setParameterWithUInt(self.__fg, s.FG_SENDSOFTWARETRIGGER, 1, self.__port)
if (err != 0):
	print(f"Trigger failed {err}: {s.Fg_getLastErrorDescription(self.__fg)}")
```