basler/pypylon #645: Failed to discover GigE device

Marco0034, 2023-08-29:
Hello, 
I have an issue to connect to my ace2 camera out of a virtualBox ubuntu environment. I am able to ping the device but get the following error: 

Traceback (most recent call last):
  File "...", line 117, in <module>
    cam = pylon.InstantCamera(tlf.CreateDevice(cam_info))
  File "...", line 1632, in CreateDevice
    return _pylon.TlFactory_CreateDevice(self, *args)
_genicam.RuntimeException: Failed to discover GigE device '10.42.230.185:3956' : RuntimeException thrown (file 'BaslerGigETl.cpp', line 680)

#Ping camera
```
hostname = "10.42.230.185" #example
response = os.system("ping -c 1 " + hostname)
#and then check the response...
if response == 0:
  print(f"{hostname} is up!")
else:
  print(f"{hostname} is down!")
```

# Connect to camera 
```
tlf = pylon.TlFactory.GetInstance()
tl = tlf.CreateTl('BaslerGigE')
cam_info = tl.CreateDeviceInfo()
cam_info.SetIpAddress('10.42.230.185')
cam = pylon.InstantCamera(tlf.CreateDevice(cam_info))
```

Thank you for your help! 
Kind regards 
Marco

SMA2016a, 2023-08-30:
try this.  Did you set the network device to NAT or Bridge mode?  It would be better if you could set in bridge mode, so the camera can be found via unicast discover command.

However test this code:

hostname = "10.42.230.185"
tlf = pylon.TlFactory.GetInstance()
ptl = pylon.TlFactory.GetInstance()
empty_camera_info = pylon.DeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', hostname)
camera_device = tlf.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)

SMA2016a, 2023-09-03:
set the virtualBox  in braidge mode and make sure that you have disabled the IP filter as written in pylon's README.

if then camera not visual, create a tcpdump  file in VirtualBox and upload   it.  While capturing the log, open ipconfig tool and press refresh button