basler/pypylon #646: Linux PATH prepended using semicolon not colon

tommyyliu, 2023-08-29:
pylon or pypylon prepends $PATH with /opt/pylon/lib/pylonCXP/bin;/opt/pylon/lib/pylonCXP/genicam/bin/Linux64_x64;
This is a Windows syntax convention that is invalid on Linux. This clobbers the path making the previous first item on $PATH invalid, while failing to properly add the CXP paths to PATH.

This issue presents itself when Pylon Viewer is installed, but does not if it isn't. Otherwise, it seems easy to reproduce across a variety of versions.

`from pypylon import pylon
import os

if __name__ == '__main__':
    transport_layer = pylon.TlFactory.GetInstance()
    print(os.popen("echo $PATH").read())
    devices = transport_layer.EnumerateDevices()
    print(os.popen("echo $PATH").read())`

tommyyliu, 2023-08-30:
No, this is just by running the provided repro script provided in the initial post.

First print will give the normal path.
Second print will give the path with the prepended string.

thiesmoeller, 2023-08-30:
Thanks for reporting this. 
We'll check this in the CXP drivers

thiesmoeller, 2023-08-31:
code in https://github.com/basler/pypylon/blob/270ab28bfccb513e4fae43870e027f65b7fde4c5/src/pylon/pylon.i#L228 was initially developed for linux only... will fix

thiesmoeller, 2023-08-31:
which pypylon / pylon version do you use?

thiesmoeller, 2023-08-31:
root cause has been found and will be fixed in next release. 
Please check if this workaround works for you:

```python
import os
import pypylon.pylon as pylon

def protect_first_path_element_from_prepend_with_semicolon():
    # workaround to keep the first element in PATH
    # a current issue in the CXP GenTL Producer
    # renders the first PATH element unusable

    path = os.environ["PATH"].split(":")
    fixed_path = [path[0]] + path
    os.environ["PATH"] = ":".join(fixed_path)

if __name__ == "__main__":

    protect_first_path_element_from_prepend_with_semicolon()

    transport_layer = pylon.TlFactory.GetInstance()
    print(os.popen("echo $PATH").read())
    devices = transport_layer.EnumerateDevices()
    print(os.popen("echo $PATH").read())
    for d in devices:
        print(d)

```

tommyyliu, 2023-08-31:
@thiesmoeller , thanks for the fast response. We have already put in workaround, just wanted to flag the issue.

closed 2023-08-31 by tommyyliu