basler/pypylon #647: Flush queue after calling `StopGrabbing`

zacharynevin-stemcell, 2023-09-01:
Hi there,

There seem to be a lot of different variables related to queueing of images.  There are variables such as `MaxNumQueuedBuffer`, `MaxNumBuffer`, and `OutputQueueSize`, and I am not clear on the difference between them.

For context, I want to do an acquisition with the `GrabStrategy_OneByOne` strategy and use an `ImageEventHandler` to handle conversion and saving to disk.

When I call `pycam.StopGrabbing()`, I want to basically "flush" the remaining items in the queue. 

For example, if I am running the camera at 60fps for 10 seconds and I stop the acquisition, I still want to save any existing items in the queue instead of discarding. Additionally, I want to ensure that even if the `ImageEventHandler.onImageGrabbed` function can't keep up with the framerate, that I don't lose any frames.

SMA2016a, 2023-09-02:
All parameters are explained on this documentation page.

https://docs.baslerweb.com/pylonapi/net/T_Basler_Pylon_PLCameraInstance

I would say, that your setup loses images since writing image to the hard disk takes longer that 16ms.

try to set MaxNumBuffer   = 500

thiesmoeller, 2023-09-02:
.. and call StopGrabbing from your OnImageGrabbed callback. Then you'll know the exactly when all images have been processed