basler/pypylon #648: Is there an alternative way of GetArray?

MrDanyBrown, 2023-09-04:
Hello everyone, I am receiving data from the camera using the burst frame feature and software triggering. When append images to the list with the help of the GetArray function(return numpy array), it takes about 10ms in each frame. This shows that I can get a maximum of 100 fps per second. 

As an alternative way, I create a pylon.Image object and attach the grabresult object to this object and add it to the list, but I see that I need to release the objects in the list in a different process. What would be the most suitable method for this job? I would appreciate it if you could share a suggested strategy.

```

`import os
import time
import cv2
from pypylon import pylon

class ImageEventPrinter(pylon.ImageEventHandler):
    def __init__(self):
        super(ImageEventPrinter, self).__init__()
        
        self.frames= []
        
    def OnImagesSkipped(self, camera, countOfSkippedImages):
        print(countOfSkippedImages, " images have been skipped.")

    def OnImageGrabbed(self, camera, grabResult):
        print(grabResult.GrabSucceeded())
        if grabResult.GrabSucceeded():
            print("Image Shape Retrieved from Software Trigger Image ", grabResult.ImageNumber)

            img = pylon.PylonImage()
            img.AttachGrabResultBuffer(grabResult)
            
            self.frames.append({"Image Number":grabResult.ImageNumber, "img":img}) #  grabResult.GetArray() -> 10 ms
            
            grabResult.Release()
        else:
            print("Error: ", grabResult.GetErrorCode(), grabResult.GetErrorDescription())
            

tl_factory = pylon.TlFactory.GetInstance()
cam = pylon.InstantCamera(tl_factory.CreateFirstDevice())

print('Start acquiring')

image_handler = ImageEventPrinter()

cam.Open()

cam.MaxNumBuffer = 10
cam.TriggerSelector = "FrameBurstStart"
cam.TriggerMode = "On"
cam.TriggerSource = "Software"
cam.TriggerActivation = "RisingEdge"
cam.PixelFormat = "BGR8"
cam.ExposureTime = 10000
cam.AcquisitionBurstFrameCount = 10

print("Trigger mode: ", cam.TriggerMode.Value)

cam.RegisterImageEventHandler(image_handler, pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
cam.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera) #First In First Out Mode

cam.ExecuteSoftwareTrigger()
time.sleep(2)

"This part must because MaxNumBuffer is reached"

for idx, frame in enumerate(range(len(image_handler.frames))):
    cv2.imwrite(os.path.join("imgs",str(idx) + ".png"), image_handler.frames[idx]["img"].Array)
    image_handler.frames[idx]["img"].Release()
    
    
image_handler.frames = []

cam.ExecuteSoftwareTrigger()
time.sleep(2)

cam.DeregisterImageEventHandler(image_handler)
cam.StopGrabbing()
cam.Close()
print("Camera Closed")

for idx, frame in enumerate(range(len(image_handler.frames))):
    cv2.imwrite(os.path.join("imgs",str(image_handler.frames[idx]["Image Number"]) + ".png"), 
                image_handler.frames[idx]["img"].Array)
    image_handler.frames[idx]["img"].Release()
    ``
```

thiesmoeller, 2023-09-05:
What is the ROI of your camera? And which model do you use?

It might be that transferring width*height*3(bgr) takes 10ms on a 1gig network?

MrDanyBrown, 2023-09-05:
> What is the ROI of your camera? And which model do you use?
> 
> It might be that transferring width_height_3(bgr) takes 10ms on a 1gig network?

Our camera is acA4112-30uc.  Since it is a 12 MP camera, it is normal to receive 10 ms of data, but I don't want to wait for the camera to receive data. For this reason, does it make sense to increase the camera buffer size and release this buffer in a different process?

thiesmoeller, 2023-09-05:
Why not use a queue in the ImageEventHandler class.
You push in a new frame and pop frames in main e.g. to save them to disk

That way you don't have to sleep in main ( if this is your problem?)

MrDanyBrown, 2023-09-05:
> 
Yes, my problem is briefly this way, I'm planning to use a similar queue structure. The only thing that confuses me is that I don't know exactly what to add in queue. Do you think it would make sense if I added pylon image object it as below?

`
img = pylon.PylonImage()

img.AttachGrabResultBuffer(grabResult) 

queue.put(img)
`
I will also pop this queue with a different process and release again.  Is this way sense ?

thiesmoeller, 2023-09-05:
As you are using openCV for saving you can add the result of GetArray to the queue. Which is an ndarray that openCV understands.

Although most memory efficient would be to save directly from the callback handler. 
You just need enough (MaxNumBuffers) to handle the case that saving is not fast enough and you have to buffer incoming frames

MrDanyBrown, 2023-09-06:
> As you are using openCV for saving you can add the result of GetArray to the queue. Which is an ndarray that openCV understands.
>
> Although most memory efficient would be to save directly from the [...]

Do you have a basic sample code for this scenario? I want to ask you a another concern;

The camera writes the data directly to the internal relevant buffer, and the ImageEventHandler performs the buffering release. 
If this is true why do we add buffer data in Image EventHandler class again? By keeping the buffer size high, we can save image in the ImageEventHandler. Do you think this thought is correct?

Thanks for your helping

thiesmoeller, 2023-09-07:
Yes this thought is correct.

If you e.g. we to capture 100 frames a save them to filesystem., but your save routine is too slow..
Just tell pylon to use 100 buffers

Normally your frame Processing should take lease time than the frame time. In that case you can live with.a small number of buffers and just circulate them around

MrDanyBrown, 2023-09-07:
> Yes this thought is correct.
>
> If you e.g. we to capture 100 frames a save them to filesystem., but your save routine is too slow.. Just tell pylon to use 100 buffers
>
> Normally your frame Processing [...]

Thank you for your helpilng this issue can be closed

closed 2023-09-07 by MrDanyBrown