basler/pypylon #649: using Multithread to start and close usb camera, the other process can't call camera.

Septembit, 2023-09-07:
Hello, I tried to use Multithread to run camera, but when I close this thread, and try to use the other process to call camera, it failed but no error message. Do anyone have the same issue?
`self.camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())`

thiesmoeller, 2023-09-07:
It sounds as if your multithreaded first process is still holding a reference to your camera.

How do you "close your camera"?

Septembit, 2023-09-07:
Thanks for your replying.
I used this code to close camera.
`self.camera.StopGrabbing()
self.camera.Close()`

thiesmoeller, 2023-09-07:
If there is really no thread holding a reference to the camera which prevents unloading

You can detach and destroy the device:

```python
dev = self.camera.DetachDevice()
pylon.TlFactory.GetInstance().DestroyDevice(dev)
```

This needs latest PyPylon 2.3.x.

Septembit, 2023-09-07:
OK, thank you, I will have a try.