basler/pypylon #653: Can't Sync Camera and NIC with PTP (IEEE1588)

AlexRoss-WHS, 2023-09-13:
I am trying to test the PTP syncronisation of the acA2440-20gc GigE cameras we use. I am using ptp4l on a linux machine to set up PTP for the port/interface the single camera I am testing with is on. The problem is that camera and PC don't seem to recognize each other when is comes to PTP. The ptp4l output ends with `ptp4l[2751.934]: port 1: assuming the grand master role`. When I check the `GevIEEE1588Status` of the camera it says `Master`. So both the camera and the NIC assume the role of master and no synchronisation takes place. I can ping the camera and connect to it just fine over that same interface.
Am I missing something here? How do I get the camera to sync with the PHC in my NIC?

SMA2016a, 2023-09-15:
have you seen this?
https://github.com/basler/pypylon/issues/603

AlexRoss-WHS, 2023-09-18:
I managed to get it working. The problem was the config I used. Apparently it does not work with the gPTP config used in the linuxptp documentation examples.

closed 2023-09-18 by AlexRoss-WHS

AIGR-sw, 2024-03-25:
@AlexRoss-WHS How did you manage to solve it? I have four cameras connected to a 4-port Ethernet card of a Linux PC. How do I have to configure the linuxptp daemon?