basler/pypylon #654: AccessException on cam.SequencerSetNext.GetValue()

SpangeJ, 2023-09-13:
Hello,

I have an application running three cameras using Sequencer.
It is 2x acA1300-200um and 1x a2A5320-23ucPRO  (FW 2.3.0 which supports Sequencer)

In my code I can access cam.SequencerSetNext.GetValue() for my first two acA cameras, but when I try to run it on the a2A I get the following error:

`Traceback (most recent call last):
  File "<input>", line 1, in <module>
    cam.SequencerSetNext.GetValue()
  File "/opt/optoscale/venvs/fishrecording/lib/python3.6/site-packages/pypylon/genicam.py", line 2162, in GetValue
    return _genicam.IInteger_GetValue(self, Verify, IgnoreCache)
_genicam.AccessException: Node is not readable. : AccessException thrown in node 'SequencerSetNext' while calling 'SequencerSetNext.GetValue()' (file 'IntegerT.h', line 149)`

I know the object is there, see (this is bpdb):
[image]

Why can I not access the node?

SMA2016a, 2023-09-15:
have you set configuration mode to on?

 camera.Open()
 camera.SequencerConfigurationMode.SetValue('On');
 val = camera.SequencerSetNext.GetValue()
 print(val)

SpangeJ, 2023-09-15:
@SMA2016a thank I'll try this out.
So I have a verry large script, but this is the essence:
```
Class Camera:
def connect(self):
# Sequencer mode
    logger.debug('SequencerMode On! Sequence length: {}'.format(self.seq_len))
    cam.GainAuto = "Off"
    cam.ExposureAuto = 'Off'
    cam.SequencerMode = 'Off'
    cam.SequencerConfigurationMode = 'On'
    for j in range(self.seq_len):
        cam.SequencerSetSelector = j
        cam.ExposureTime = exposure[j]
        cam.Gain = gain[j]
        cam.SequencerSetSave.Execute()
    cam.SequencerSetNext = 0
    cam.SequencerSetStart = 0
    cam.SequencerSetSave.Execute()
    cam.SequencerConfigurationMode = 'Off'
    cam.SequencerMode = 'On'
    logger.info('Sequencer camera {}: exposure: {}, gain = {}'.format(i, exposure, gain))

def ensure_start_of_sequence(self):
    for i, cam in enumerate(self.camera):
        if cam.SequencerSetNext.Value != 0:
            logger.error('Cam {}: Sequencer out of sync!. Restart SW!'.format(i))

```

I.E. I connect to the cameras, then I send N trigger signals and acquire N images, and since the length of my sequencer is N, I expect to have looped through the set on all my cameras. However, I would like to verify that by checking.
And this works if my cameras are `acA`, but not `a2A`.

Are you telling me that I have to alter `ensure_start_of_sequence` to:

```
def ensure_start_of_sequence(self):
    for i, cam in enumerate(self.camera):
        if 'a2A' in cam.DeviceModelName.Value:
            cam.SequencerConfigurationMode.SetValue('On');
        if cam.SequencerSetNext.Value != 0:
            logger.error('Cam {}: Sequencer out of sync!. Restart SW!'.format(i))
        if 'a2A' in cam.DeviceModelName.Value:
            cam.SequencerConfigurationMode.SetValue('Off');
```
NB! Remember, that at this stage `cam.SequencerMode = 'On'`

thiesmoeller, 2023-09-15:
Ace2 has some slightly more modern sequencer config.
( we updated to latest genicam spec )

We'll provide a pypylon sample for ace2 sequencer in the next week

SpangeJ, 2023-09-20:
@thiesmoeller 

I have a reproducible bug now that I would like your help with,
How to reproduce:

1. My application crashes on the error explained in this thread `_genicam.AccessException: Node is not writable : AccessException thrown in node 'SequencerConfigurationMode' while calling 'SequencerConfigurationMode.FromString()'`
2. I re-run the script and then I get the following error:
`cam.Open()  File /pypylon/pylon.py, line 3308, in Open return _pylon.InstantCamera_Open(self)
_genicam.RuntimeException: Could not apply configuration. Pylon::GenericException caught in OnOpened method msg=Node is not writable : AccessException thrown in node 'TriggerMode' while calling 'TriggerMode.FromString()' (file 'ValueT.h', line 85) : RuntimeException thrown (file 'AcquireContinuousConfiguration.h', line 76)`

I am able to resolve this, but that is a manual job where I open Pylon Viewer, connect to the cameras and select "Device control" --> "Device reset" --> "Execute". The camara disappears and reappears after about five seconds.
How do I resolve this error with `.Open()` in a script?

thiesmoeller, 2023-09-20:
The camera state stays on USB and GEV even if you shutdown the application.

I assume the camera ist still in sequencer mode.

Just disable it

SpangeJ, 2023-09-20:
@thiesmoeller Also, Device reset is buggy
I have three cameras, but after Device Reset, only two comes back.
[image]
Rebooting the system makes my three cameras re-appear:
[image]

thiesmoeller, 2023-09-20:
How do the cameras get their IP address ?

thiesmoeller, 2023-09-20:
What is your platform and OS version ?
What does dmesg say?

SpangeJ, 2023-09-20:
@thiesmoeller I can reproduce the issue, not consistently, but If I try `Device Reset` (after connecting) in Pylon Viewer multiple times I can reproduce it.

[image]
This is interesting:
[image]

I ran `dmesg -wH` [dmesg.txt](https://github.com/basler/pypylon/files/12672196/dmesg.txt) is the output.

PylonViewer:
[image]

OS:
[image]

thiesmoeller, 2023-09-21:
https://github.com/basler/pypylon-samples/blob/main/notebooks/Ace2_USB_hdr_exposure_bracketing_using_sequencer.ipynb

SpangeJ, 2023-09-21:
@thiesmoeller thank you I'll have a look at this ASAP

SpangeJ, 2023-09-21:
Just by reading the documentation it looks like
`sequence_id_chunk = res.ChunkDataNodeMap.ChunkSequencerSetActive.Value` will solve the issue I address here.
I will test it out within, and you will here from me next week.

Thank you verry much, @thiesmoeller

SpangeJ, 2023-09-26:
@thiesmoeller Thank you for the documentation, I can confirm what the issue is resolved.

closed 2023-09-26 by SpangeJ