basler/pypylon #656: blaze-102 and ace acA1300-75gc - PTP offset < 100 ns but timestamp difference 37 s

mxmahmoud, 2023-09-18:
No matter what i am doing. The timestamp of the blaze-102 never comes close the timestamp of the other devices,
although it is claiming to be only off by 40 ns according to the PTP protocol. Issue is reproducable with harvester and the pypylon library.

Output:

```
    Blaze:  1695028477400151139
    Ace:  1695028514444967582
    Difference: -37044816443
    Blaze:  1695028477450167548
    Ace:  1695028514494967582
    Difference: -37044800034
    Blaze:  1695028477500172318
    Ace:  1695028514544967582
    Difference: -37044795264
    Blaze:  1695028477550159606
    Ace:  1695028514594967566
    Difference:  -37044807960
    Blaze:  1695028477600191496
    Ace:  1695028514644967582
    Difference: -37044776086
```

Synchronization check via offset:

```
    cam_0_model_name = cams[0].DeviceInfo.GetModelName()
    cam_1_model_name = cams[1].DeviceInfo.GetModelName()
    
    while True:
        cams[0].GevIEEE1588DataSetLatch()
        cams[1].PtpDataSetLatch()
        cam_0_offset = abs(cams[0].GevIEEE1588OffsetFromMaster.GetValue())
        cam_1_offset = abs(cams[1].PtpOffsetFromMaster())
        print(f"Sensor offset:\n{cam_0_model_name}: {cam_0_offset} ns\n{cam_1_model_name}: {cam_1_offset} ns")
        if cam_0_offset < SYNC_TOLERANCE_NS and cam_1_offset < SYNC_TOLERANCE_NS:
            break            
        sleep(0.1)
```

Timestamp Check blaze and ace:
`  timestamp = buffer.timestamp`

SMA2016a, 2023-09-23:
did you set the frame rate of both camera to the same value?
do you run ace camera at Syncfreerun mode?

SMA2016a, 2023-09-23:
are the cameras connected to a network switch?  if yes, try another one.
is one of the camera run as PTP master?  I think this is not the case.

SMA2016a, 2024-07-28:
since no answer for a long time, I assume that it is no longer current one

closed 2024-07-28 by SMA2016a