basler/pypylon #657: How can I capture images differently from the trigger when Line1 is in trigger mode?

mehmetgencg, 2023-09-19:
I am currently using Basler cameras with the Pypylon application, and the cameras are operating in Line1 trigger mode. However, I would like to capture images from these cameras at specific intervals, in addition to the Line1 trigger mode like GrabOne. I am looking for guidance on how to achieve this.

mehmetgencg, 2023-09-19:
Thank you for the answer. I followed the method you said, but speed is important to me and I don't want to constantly change the camera configuration. Is there any other way to do this? like one time shot

thiesmoeller, 2023-09-19:
It is just:

TriggerSource = Software
Triggersoftware execute
TriggerSource = Line1

This costs you only in the millisecond range.

As the full camera and grab engine stays active, it is faster than GrabOne

mehmetgencg, 2023-09-20:
Thank you for your response

closed 2023-09-20 by mehmetgencg