basler/pypylon #658: How to use SW trigger to acquire image with undefined image height

MYCL94, 2023-09-22:
Hello all,

first of all I tried to understand the usage of the methods by reviewing the samples and searching in this area also. 
The nearest issue I did find was this post:  https://github.com/basler/pypylon/issues/503

### What do I want to achieve? 

1. Set the line rate of the camera to a defined value
2. Start acquisition
3. Stop acquisition
4. Get the complete image with n (numer of lines) from the camera

I have also a setup with complete hardware trigger (frame start, end and line trigger), which is already working as described above. But achieving the same with pyplon was not possible, whereas I don't want to trigger each line separately in this case since the linerate is fix. 

### My solution right now: (works BUT after 5-7 days the camera freezes out of nothing and only repowering helps)

No Software Trigger is set in settings, which I find curious but it works as desired.

```
self.camera.set_width(8192)
self.camera.set_height(1)
```

```
def trigger_start(self) -> None: 
  if self._camera.IsGrabbing():  return
  self._acquisition_triggered = True
  self._camera.StartGrabbing(pylon.GrabStrategy_OneByOne)
  self._camera.ExecuteSoftwareTrigger()
  Thread(target=self._trigger_start_thread, args=(self._all_buffers_retrieved_event, 0)).start()
```

```
def _trigger_start_thread(self) -> None:
while True:
  grab = self._camera.RetrieveResult(int(time_out_ms), pylon.TimeoutHandling_Return)
  if grab.IsValid() and grab.GrabSucceeded():  
      self._grab_list.extend(grab.GetArray())
  elif not self._acquisition_triggered:
      grab.Release()
      break
```

After some time I stop the acquisiton from a different thread

```
def trigger_start(self) -> None: 
  self._camera.AcquisitionStop.Execute()
  self._camera.StopGrabbing()
```

and collect the image with convert image. This is the example which works for 5-7 days until the camera freezes. My guess would be there is a much better practice to acquire the n-lines until a software stop is received. I have read about background loops and events but I haven't been successfull since I want to have partial closing frames.

I also tried setting the image height up to 3000 pixels.
But the last part of the image was missing after stopping the acquisition. I don't understand why the acquisition didn't complete the last image but threw it away instead of finishing and putting it into the buffer. Also the feature partial closing frames was not helpful with this. 

I hope you have some sort of example for this type of an application since most of the examples only containn area scan cameras.

Thanks a lot!

MYCL

thiesmoeller, 2023-09-22:
What is the camera model ?

MYCL94, 2023-09-22:
> What is the camera model ?

raL8192-12gm - Basler racer

SMA2016a, 2023-09-24:
I think you wanted set higher image height in this case.

self.camera.set_height(1)

would you explain once more what you exactly wanted to do?  Do you want to send one trigger and camera should grab some specific count of lines in freerun mode?

[I have also a setup with complete hardware trigger (frame start, end and line trigger), which is already working as described above. But achieving the same with pypylon was not possible, whereas I don't want to trigger each line separately in this case since the linerate is fix]

MYCL94, 2023-09-24:
I want to start grabbing with a fix linerate and stop whenever I want (time is not fix). So as a result I could have an image with 3000 pixel height or even 9000 pixel. 

Example:
When I set the image height to 3000 pixels with a linerate of 2000 
and acquire for about 2.5 seconds. I should have 5000 pixels in two images or maybe 6000 pixels because one image is set to 3000 pixels. But the acquisition result is only one image containing 3000 pixels. The last frame seems to get lost. 
Thats why I would like to find the best practice for software trigger acquisition.

SMA2016a, 2024-01-08:
@MYCL94  I'm reaching out regarding the status of a particular case on our GitHub repository.

Upon reviewing the case, it seems that the issue or question addressed may no longer be relevant or current. To maintain the clarity and organization of our repository, I kindly request your assistance in updating the status of this case to "Closed" if it has been resolved or if it's no longer applicable

closed 2024-09-19 by MYCL94