basler/pypylon #660: Counter reset in pypylon

Marco0034, 2023-09-23:
I use an Ace2 camera with a HTL-encoder signal as a trigger. The counter value in the camera is used to parameterize at which encoder-step the camera shot is triggered. 

In the pylon viewer everything works fine, if I reset the counter value before starting a continuous shot. 

For the implementation in the real use case I use the pypylon api. 

Is there a way in pypylon to achieve a counter reset before start grabbing? 

I appreciate the help 

Kind regards

SMA2016a, 2023-09-24:
would you tell me which parameters you are setting up i pylon viewer? Normally, you can take the node name of parameters that you are setting up?

Marco0034, 2023-09-25:
[image]

SMA2016a, 2023-10-10:
simply mark the parameter that you want to access and copy the c++ code from the document pane in pylonviewer.

if this does not work, let me know

Marco0034, 2023-10-10:
I marked the Reset value. 
[image]
Where can I find the document pane with the related python code. I could not find it.

SMA2016a, 2023-10-15:
[image]

HighImp, 2023-10-17:
The Node CounterReset is an ICommand, and the correct way to fire a command is:

camera.<node_name>.Execute()

so in your case:

camera.CounterReset.Execute()