basler/pypylon #663: When grabbing images and camera gets disconected i alway get last grab array with shape (0,1920)

hlacikd, 2023-09-28:
I have simple grabbing look like this :

```
  while camera.IsGrabbing():
        # Wait for an image and then retrieve it. A timeout of 500 ms is used.
        with camera.RetrieveResult(500, pylon.TimeoutHandling_Return) as grab:
            # Accessing the image buffer
            image = grab.GetArray()
            # post-processing
            frame.put(image)

```

I am using a2A1920-51gcPRO.

When i disconnect camera from ethernet what i expect is timeout of 500ms is handled by calling return from function. This happens correctly , however all the time, the "last" image is recieved with grab.GetArray() with dims (0,1920)! and only after return is called.

This causes error in my code due to wrong frame dimensions (should be (1080,1920).

I have workerounded this issue by adding this simple if 

```
  while camera.IsGrabbing():
        # Wait for an image and then retrieve it. A timeout of 500 ms is used.
        with camera.RetrieveResult(500, pylon.TimeoutHandling_Return) as grab:
            # Accessing the image buffer
            image = grab.GetArray()
            if all([dim > 0 for dim in image.shape]):
                 # post-processing
                 frame.put(image)
             else:
                  logger.warning(f"{self.key}> Frame dimension are {frame.shape}")
```
which is workaround around this issue , but i need to understand why is it happening and how to make it not happening --- is this a bug in pypylon?

thiesmoeller, 2023-09-28:
Your code currently only covers the happy path.
The last frame when you unplug your device might already be in receive by the grab engine. Every frame that is received will be delivered, but with error flags set in this case.

See the main example in readme https://github.com/basler/pypylon/blob/26cd419f708542b64a39f80392d21c753691f6a2/README.md?plain=1#L39

Which shows how to first check if the GrabResult is ok, before accessing the data

hlacikd, 2023-09-28:
@thiesmoeller thanks, adding grabResult.GrabSucceeded() condition solved the issue.

closed 2023-09-28 by hlacikd