basler/pypylon #668: Obtaining image data in the packed format

danielkonecny, 2023-10-12:
Hello,
I'd like to save some disk space and also improve performance of my recording app by trying to save image data in packed format. I am already saving color images in raw BayerRG (so single channel), but whenever I record 10 or 12-bit images, I have to save them in 16-bit integers and therefore lose some disk space and with that also app performance (disk writing time).

I know my ace 2 USB cameras support also "packed" format for 10 and 12-bit images but whenever I access the raw image data via grab_result.Array (in OnImageGrabbed method in ImageEventHandler, to be more precise), they are represented as np.int16 with the correct shape of the raw image.

My question is how can I really obtain the image in packed format, since right now the image data seems to be the same for "packed" and "non-packed" pixel format. It's still a 16-bit integer with 6 (for 10-bit image) or 4 (for 12-bit image) most significant bits equal to zero. I would prefer to avoid pylon's converters to save some CPU time, if it's possible. But I don't know if the camera itself can produce the data in packed format or if there is some conversion to be done after grabbing. 

Thanks in advance!
Daniel

thiesmoeller, 2023-10-12:
GetArray() unpacks the data for you, as numpy can't handle packed arrays.

If you want to save the raw data from the camera ...Use

GetBuffer()

danielkonecny, 2023-10-13:
Thank you for the quick response.

I am getting the buffer now, but I am struggling with parsing it. I have 2 problems that I am observing right now:

1. The image data has resolution (shape) 1390x2472 and 12-bit depth, which means 1390 * 2472 = 3,436,080 pixels, that is 3,436,080 * 12 = 41,232,960 bits in total. If these were separated into bytes, they should have size of 41,232,960 / 8 = 5,154,120. But the buffer size (according to len() and also GetBufferSize) is 5,154,184 bytes. That means there are additional 64 bytes in the buffer. Where do they come from and what is their purpose? I have to note that I am using chunk mode with timestamp, exposure time and gain activated. Is it possible that these values are in the buffer as well? The only specification about chunks I found is this [this](https://docs.baslerweb.com/data-chunks) one, does the image on this page display the whole buffer?
2. When I compare first 12 bits of the buffer with the first pixel of the unpacked image (from GetArray), the values differ. Am I accessing an incorrect part of the buffer? Is there some offset that I have to apply?

thiesmoeller, 2023-10-13:
Yes it's the chunk data at the end of the buffer.
Image on your camera will always be starting at offset 0 of the buffer.
Quick solution is to save ceil(Width*Height*BitPerPixel/8) from the buffer.

danielkonecny, 2023-10-13:
Thanks, that makes sense.

I also managed to get the correct "packed" structure of the buffer, for future readers, this print should guide you, when using 12-bit packed format:
`print(f"{grab_result.Buffer[1] % (2**4):04b}{grab_result.Buffer[0]:08b} = {grab_result.Array[0][0]:012b}")`

Last question on my side, little bit out of scope but it follows this topic: Is there a standardized way how to unpack the buffer? Can pylon Converter be used? I haven't found any way how to do this in pure Python without needing to shift and mask bits for each pixel (very inefficient IMO). Standard mathematical libraries like NumPy also don't support this, as far as I know.

HighImp, 2023-10-17:
You can call: grab_result.GetPixelType() to get a integer number of the pixel type according to [PFNC](https://www.emva.org/wp-content/uploads/GenICamPixelFormatValues.pdf) Since this is standardized, you can use a local lookup dictionary for the bit depth.

closed 2023-10-20 by danielkonecny