basler/pypylon #669: EnumerateDevices() doens't work with sudo

barbiomalefico, 2023-10-13:
I have a basler camera connected to the MIPI of an nvidia jetson nano.
When I run python using my normal user I'm able to enumarate the connected device:
```
#python
Python 3.8.10 (default, May 26 2023, 14:05:08)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
>>> from pypylon import genicam
>>> pyinst = pylon.TlFactory.GetInstance()
>>> pyinst.EnumerateDevices()
(<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x7f9867e7b0> >,)
>>> exit()
```

When I do the same using sudo, the device can't be enumerated. 

```
#sudo su
#python
Python 3.8.10 (default, May 26 2023, 14:05:08)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
>>> from pypylon import genicam
>>> pyinst = pylon.TlFactory.GetInstance()
>>> pyinst.EnumerateDevices()
()
>>> exit()
```

thiesmoeller, 2023-10-13:
Call `sudo -E` to transfer the GenTL path environment variable

SMA2016a, 2023-10-15:
or you can also set persistent environment variable to all users:    GENICAM_GENTL64_PATH=/opt/dart-bcon-mipi/lib/

if the problem is solved, please close this thread.

barbiomalefico, 2023-10-16:
The problemi is solved

closed 2023-10-16 by barbiomalefico